kristinemlarson/gnssrefl #24: invsnr -knot_space type error

timdittmann, 2022-09-20:
A user has identified that the `invsnr -knot_space` argument throws a type error.

E.G.
`invsnr tgmx 2022 55 ALL -knot_space 2.0
`
results in:
```
Traceback (most recent call last):
  File "/usr/local/bin/invsnr", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/gnssrefl/invsnr_cl.py", line 292, in main
    invsnr(**args)
  File "/usr/local/lib/python3.8/dist-packages/gnssrefl/invsnr_cl.py", line 282, in invsnr
    spline_functions.snr2spline(station, year, doy, azilims, elvlims, rhlims, precision, kdt, signal=signal,
  File "/usr/local/lib/python3.8/dist-packages/gnssrefl/spline_functions.py", line 747, in snr2spline
    knots = np.linspace(gbase + int(kdt/2), gbase + numdays*86400 - int(kdt/2), int(numdays*86400/kdt))
TypeError: unsupported operand type(s) for /: 'str' and 'int'
```

kristinemlarson, 2022-09-20:
This bug is fixed. The knot parameter is defined as an integer and if someone inputs a float, they will be notified not to input a float.

closed 2022-09-20 by kristinemlarson

timdittmann, 2022-09-21:
[Fixed.](https://github.com/kristinemlarson/gnssrefl/commit/af88eafb4ce6a71d01d33b7862549e39d11db8a3).