kristinemlarson/gnssrefl #31: Compilation failed with python3.10 but successful with 3.9

pmlefeuvre, 2022-11-03:
Hi,

I have tried to install gnssrefl on Fedora following your **README_install.md** and based on its github version. 
Using Python 3.10.7, the compilation of gnssrefl fails as seen in the output below.

The solution was to create my environment using Python3.9 and the rest worked.
`sudo dnf install python3.9` for Fedora or alternatively `sudo apt-get install python3.9`
`python3.9 -m venv env` to create your environment
`source env/bin/activate` to activate it

then
`pip install wheel`
`pip install .`  compiles gnssrefl smoothly

Perhaps someone else got that problem?

Cheers,

Pierre

```
(env) [xxx@xxx gnssrefl]$ pip install wheel
Collecting wheel
  Downloading wheel-0.37.1-py2.py3-none-any.whl (35 kB)
Installing collected packages: wheel
Successfully installed wheel-0.37.1
WARNING: You are using pip version 21.3.1; however, version 22.3 is available.
You should consider upgrading via the '/home/xxx/gnssrefl/env/bin/python3 -m pip install --upgrade pip' command.
(env) [xxx@xxx gnssrefl]$ pip install .
Processing /home/xxx/gnssrefl
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting astropy
  Downloading astropy-5.1.1-cp310-cp310-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (11.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 11.0/11.0 MB 30.8 MB/s eta 0:00:00
Collecting scipy
  Downloading scipy-1.9.3-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (33.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 33.7/33.7 MB 27.8 MB/s eta 0:00:00
Collecting requests
  Downloading requests-2.28.1-py3-none-any.whl (62 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 62.8/62.8 kB 7.4 MB/s eta 0:00:00
Collecting numpy
  Using cached numpy-1.23.4-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (17.1 MB)
Collecting progress
  Downloading progress-1.6.tar.gz (7.8 kB)
  Preparing metadata (setup.py) ... done
Collecting wget
  Downloading wget-3.2.zip (10 kB)
  Preparing metadata (setup.py) ... done
Collecting matplotlib
  Downloading matplotlib-3.6.2-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (11.8 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 11.8/11.8 MB 39.6 MB/s eta 0:00:00
Collecting pyerfa>=2.0
  Downloading pyerfa-2.0.0.1-cp310-cp310-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_12_x86_64.manylinux2010_x86_64.whl (743 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 743.2/743.2 kB 72.2 MB/s eta 0:00:00
Collecting packaging>=19.0
  Downloading packaging-21.3-py3-none-any.whl (40 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 40.8/40.8 kB 8.6 MB/s eta 0:00:00
Collecting PyYAML>=3.13
  Downloading PyYAML-6.0-cp310-cp310-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_12_x86_64.manylinux2010_x86_64.whl (682 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 682.2/682.2 kB 21.3 MB/s eta 0:00:00
Collecting python-dateutil>=2.7
  Downloading python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 247.7/247.7 kB 45.6 MB/s eta 0:00:00
Collecting cycler>=0.10
  Downloading cycler-0.11.0-py3-none-any.whl (6.4 kB)
Collecting pillow>=6.2.0
  Downloading Pillow-9.3.0-cp310-cp310-manylinux_2_28_x86_64.whl (3.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.3/3.3 MB 42.5 MB/s eta 0:00:00
[... 233 lines cut ...]
      CPU baseline  :
        Requested   : 'min'
        Enabled     : SSE SSE2 SSE3
        Flags       : -msse -msse2 -msse3
        Extra checks: none
      
      CPU dispatch  :
        Requested   : 'max -xop -fma4'
        Enabled     : SSSE3 SSE41 POPCNT SSE42 AVX F16C FMA3 AVX2 AVX512F AVX512CD AVX512_SKX AVX512_CLX AVX512_CNL AVX512_ICL
        Generated   : none
      INFO: CCompilerOpt.cache_flush[857] : write cache to path -> /home/xxx/gnssrefl/build/temp.linux-x86_64-3.10/ccompiler_opt_cache_ext.py
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for gnssrefl
  Building wheel for progress (setup.py) ... done
  Created wheel for progress: filename=progress-1.6-py3-none-any.whl size=9632 sha256=233716da174c3d06de91d695fea9a93666e0406221ef174c7581bc48a6089410
  Stored in directory: /home/xxx/.cache/pip/wheels/c6/d5/4b/a9bd8a1bc5ebcf4c9cfe35795a5d859e15492f338d2d1a623a
  Building wheel for wget (setup.py) ... done
  Created wheel for wget: filename=wget-3.2-py3-none-any.whl size=9675 sha256=fc904d59e6f40154532ac2af92dcbb0ef7340ab38261b050e7d339cec4791661
  Stored in directory: /home/xxx/.cache/pip/wheels/46/78/0e/8e5e2b500f83a682c8d7e7ce820638cf99faa894a662f71cf0
Successfully built progress wget
Failed to build gnssrefl
ERROR: Could not build wheels for gnssrefl, which is required to install pyproject.toml-based projects
```

kristinemlarson, 2022-11-04:
Thank you!  We are in the middle of updating our docs - but I will add this information to our README as soon as possible.  And I should probably put a warning in the video on installing as well.  I believe this issue with 3.9 vs 3.10 is due to our reliance on the fortran bindings - and the short cut for now is to use 3.9.

kristinemlarson, 2022-11-15:
We have added a warning about the need for 3.9.  We are unable to support python >= 3.10 at this time. Thank you again for letting us know.

closed 2022-11-15 by kristinemlarson