kristinemlarson/gnssrefl #43: readthedocs docker instructions convert -- to en dash

dnowacki-usgs, 2023-01-06:
On the rendered readthedocs [docker instructions page](https://gnssrefl.readthedocs.io/en/latest/pages/docker_cl_instructions.html), the code block for starting docker converts the two hyphens in front of the `name` argument to an en dash:

[image]

After I realized the dash before `name` looked a little different than those in front of the other arguments, I inspected the original .md source and saw it should have been a double hyphen. GitHub [renders the source correctly](https://github.com/kristinemlarson/gnssrefl/blob/master/docs/pages/docker_cl_instructions.md):

[image]

Looking at the HTML source using dev tools shows this is a server-side issue and not a local rendering problem... but just in case, tested on latest Safari and Firefox on macOS 11.7.2.

[image]

I suspect this is a readthedocs setting/issue/bug but wanted to let you know that it's a potential stumbling block for those trying to run the docker image.

kristinemlarson, 2023-01-06:
Thank you for letting us know!  If I can't figure out how to fix it, I will have Tim do it as soon as he gets back from vacation.

kristinemlarson, 2023-01-07:
This issue has been fixed by using <PRE> rather than <code>.

dnowacki-usgs, 2023-01-09:
Thanks! The code block is rendering correctly for me now. However, the "please note" section above the code block is changing the double hyphen to an en dash in the alert text. At least the important part is copy-pastable now.

kristinemlarson, 2023-01-09:
sigh, i need to take that out.  so many moving pieces (fix code, commit
code, git push, and then readthedocs is
being done manually for the time being, though i think we should change
that to automatically cause i am the
one who gets stuck doing it !)  it is better than
it used to be though. for a while the changes i made in code were showing
up on github, but NOT in the readthedocs
and i could not get the guy who set up the readthedocs to believe me ...

on the plus side, after christmas my son set up ci tests and smoke tests (i
think he called them), and it actually did find a place
where i had an errant print statement that crashed a major module.  so i
would call that a win for continuous integration.

k.

-- 
Kristine M. Larson
Professor Emerita, University of Bonn
***@***.***
https://www.kristinelarson.net <http://www.kristinelarson.net>
https://gnss-reflections.org
https://github.com/kristinemlarson

Don't worry about the overall importance of the problem; work on it if it
looks interesting. I think there's a sufficient correlation between
interest and importance.
— David Blackwell