kristinemlarson/gnssrefl #44: Support GN/GL-prefixed NMEA strings

dnowacki-usgs, 2023-01-09:
I have NMEA strings from a consumer-grade USB-serial GNSS receiver. It can receive from GPS, GLONASS, and Galileo satellites but outputs sentences with a variety of prefixes. Running `nmea2snr` currently fails since it checks only for GPGGA and GPGSV, not those with other prefixes:

https://github.com/kristinemlarson/gnssrefl/blob/a60e70bda2d6f600103e4f343874db0f069a7691/gnssrefl/nmea2snr.py#L130

https://github.com/kristinemlarson/gnssrefl/blob/a60e70bda2d6f600103e4f343874db0f069a7691/gnssrefl/nmea2snr.py#L138

In testing, I was able to produce an SNR file with `nmea2snr` by changing these tests to `if b"GGA" in line:` and `elif b"GSV" in line:`

Happy to submit a PR if this seems useful, or would be grateful to see this change made in the code.

I'm including a sample NMEA output from this receiver below: 
```
$GNRMC,000000.000,A,7123.1189,N,15627.6746,W,0.03,0.00,010922,,,D*6E
$GNVTG,0.00,T,,M,0.03,N,0.06,K,D*23
$GNGGA,000001.000,7123.1189,N,15627.6747,W,2,16,0.59,20.1,M,0.3,M,,*6A
$GNGLL,7123.1189,N,15627.6747,W,000001.000,A,D*52
$GPGSA,A,3,12,31,06,17,25,19,21,32,22,04,,,2.06,0.59,1.98*02
$GLGSA,A,3,87,79,69,85,78,71,,,,,,,2.06,0.59,1.98*1C
$GPGSV,4,1,14,03,69,178,24,22,37,070,44,31,37,097,42,19,36,281,40*72
$GPGSV,4,2,14,04,31,201,21,01,28,171,20,17,27,250,34,06,25,308,43*7C
$GPGSV,4,3,14,12,23,353,49,25,22,032,45,32,12,058,42,21,08,164,17*71
$GPGSV,4,4,14,44,05,139,28,09,02,220,21*70
$GLGSV,2,1,06,79,33,073,44,78,32,009,49,71,31,238,36,69,28,073,44*69
$GLGSV,2,2,06,85,28,197,26,87,26,334,45*61
$GNRMC,000001.000,A,7123.1189,N,15627.6747,W,0.02,0.00,010922,,,D*6F
$GNVTG,0.00,T,,M,0.02,N,0.03,K,D*27
$GNGGA,000002.000,7123.1189,N,15627.6747,W,2,15,0.61,20.1,M,0.3,M,,*61
$GNGLL,7123.1189,N,15627.6747,W,000002.000,A,D*51
$GPGSA,A,3,12,31,06,17,25,19,21,32,22,,,,1.08,0.61,0.90*09
$GLGSA,A,3,87,79,69,85,78,71,,,,,,,1.08,0.61,0.90*13
$GPGSV,4,1,14,03,69,178,24,22,37,070,44,31,37,097,42,19,36,281,40*72
$GPGSV,4,2,14,04,31,201,21,01,28,171,20,17,27,250,34,06,25,308,43*7C
$GPGSV,4,3,14,12,23,353,49,25,22,032,45,32,12,058,42,21,08,164,17*71
$GPGSV,4,4,14,44,05,139,28,09,02,220,17*75
$GLGSV,2,1,06,79,33,073,44,78,32,009,49,71,31,238,36,69,28,073,44*69
$GLGSV,2,2,06,85,28,197,26,87,26,334,46*62
```

kristinemlarson, 2023-01-09:
am happy to add this if it will have both behaviors (i.e. current and
yours).
pull request is fine.

glad someone is testing that - because i got the code from makan karegar
and i don't really
have any test files anymore.  can you send me a few of your files so i have
them?  and i'll
ask makan to send me some of his too.

k.

-- 
Kristine M. Larson
Professor Emerita, University of Bonn
***@***.***
https://www.kristinelarson.net <http://www.kristinelarson.net>
https://gnss-reflections.org
https://github.com/kristinemlarson

Don't worry about the overall importance of the problem; work on it if it
looks interesting. I think there's a sufficient correlation between
interest and importance.
— David Blackwell

fgnievinski, 2023-01-10:
I'd suggest generalizing only the GGA sentence, to extract the GNSS date and time.
the GSV is best left as specific to GPS SNR data; otherwise, GLONASS SNR data would be output as if it were GPS.

full support for GLONASS would require taking care
https://github.com/kristinemlarson/gnssrefl/blob/master/docs/pages/rinex2snr.md
- satellite number (remember 100 is added for Glonass, etc)
- names for the GNSS frequencies: 1,2 are GPS L1, L2 ... 101,102 are Glonass L1 and L2

dnowacki-usgs, 2023-01-10:
Thanks for pointing that out. I updated the PR #45 to add the offset for the satellite numbers for the other systems. Note that this changes the alignment of the produced SNR file, which I did not correct for. It seems to create reasonable-looking quickLook plots. I didn't want to go too far down this rabbit hole of correcting the alignment, etc until checking in with you all.

kristinemlarson, 2023-01-10:
sorry i have been distracted today with some medical appointments and
testing
a new utility.  will give a look to see if i understand the discussions.
it has been a while
since i looked at NMEA.

-- 
Kristine M. Larson
Professor Emerita, University of Bonn
***@***.***
https://www.kristinelarson.net <http://www.kristinelarson.net>
https://gnss-reflections.org
https://github.com/kristinemlarson

Don't worry about the overall importance of the problem; work on it if it
looks interesting. I think there's a sufficient correlation between
interest and importance.
— David Blackwell

kristinemlarson, 2023-01-28:
This issue is resolved.  Let us know if you find issues with Glonass satellites.

closed 2023-01-28 by kristinemlarson

jonauq2508, 2024-03-21:
@dnowacki-usgs  I would like to know which reveicer were you using, thanks

dnowacki-usgs, 2024-03-21:
I think it was a YIC GU-902MGG-USB 

https://www.yic.com.tw/wp-content/uploads/2020/11/YIC-GU-902GG-GU-902MGG.pdf