kristinemlarson/gnssrefl #46: Add SWE module to gnssrefl

kristinemlarson, 2023-01-11:
gnssrefl is a python version of the code used in the PBO H2O project, https://gnss-reflections.org/maps.  one of the products of that project was snow depth.  We made a special effort to make a SWE product as well. We have that SWE code in matlab and it would make sense to port it to python to include it here. If someone from the cryo world would volunteer to do this port, I would happily provide the Matlab code. It is not terribly long or complicated code. It was written under the leadership of Eric Small, with an abandoned version in R written by James McCreight, and a Matlab version written by Mark Raleigh.

derekpickell, 2023-01-11:
Hi Kristine! Would love to take a look... I've been porting a lot of cryo-related Matlab code into python recently and can give this a try.

kristinemlarson, 2023-01-14:
I have found the code. It was originally written in R to analyze multiple years of hundreds of stations, and then ported to Matlab. I stripped it down so that it acts on a single station and a single year. Nevertheless, it is very clear to me that the only way this port to python will be successful is if the matlab code is first digested and improved (i.e. functions were not effectively used in the version I am giving you and the "main code" is far too long). If you are ready to digest the matlab to see what this python port would entail- I am happy to pass it on to you.

derekpickell, 2023-01-17:
Sounds to me like it needs further modularization. Will have a better idea what I'm getting myself into once I lay eyes on the existing code... I've sent you an email.

kristinemlarson, 2023-01-18:
I have placed the stripped-down code in a private github repository for now.  I have test cases so you can be sure you are getting the same answer in Matlab as in python. Anyone interested in trying to port it can contact me via email for access. Thanks, Kristine

kristinemlarson, 2023-08-19:
i don't see much interest in this from the hydrology, atmospheric, and soil moisture communities, so am moving it for now.

closed 2023-08-19 by kristinemlarson