kristinemlarson/gnssrefl #50: Soil Moisture code for sites with significant vegetation

kristinemlarson, 2023-01-20:
Our current soil moisture module (vwc) is based on the operational PBO H2O model. This worked well in the grasslands and savannahs of the western U.S. These regions are generally categorized as semi-arid. We know this model fails for regions with vegetation water content of > 2kg/m/m. Clara Chew developed a better model for these more challenging sites.  It was coded in Matlab - and could easily be ported to python.  Let me know if you are interested in tackling this port. For details on Clara's work, you can look at  https://www.kristinelarson.net/wp-content/uploads/2015/10/ChewSmallLarson2015_withoutFront.pdf

kristinemlarson, 2023-04-04:
I am consolidating this issue with the SWE issue.

kristinemlarson, 2023-10-31:
ETH might have an interest?

mnkosary, 2024-07-31:
Hello. I read Clara's work and I am so interested in this topic. I wanted to ask you, are the codes for this work available?

kristinemlarson, 2024-08-07:
I don't have this code working.

closed 2024-08-07 by kristinemlarson