kristinemlarson/gnssrefl #80: why no snr file was creatied by using brst station?

1026689430, 2023-03-07:
when type this "rinex2snr brst 2021 1" ,No SNR file was created - check logs section for additional information

kristinemlarson, 2023-03-07:
I am not sure why it is not created. The command works on my computer. Could you look in the logs folder?  Look for a filename with brst in it - and see what it says.  Also, could you make sure your environment variables are set?  i.e type printenv REFL_CODE.

kristinemlarson, 2023-03-07:
another question - does rinex2snr work for any data?  Or is the problem just with brst?

1026689430, 2023-03-07:
(env) t@t-VirtualBox:~$ rinex2snr brst 2021 1
Will seek the RINEX file from an external archive
brst  year: 2021  doy: 1 from:  all
did not find lowercase station snr file
Requested a GPS only nav file
Nav file exists online
Orbit file:  /home/t/gnss/io/2021/nav/auto0010.21n
Searching the  unavco  archive for  brst
https://data.unavco.org/archive/gnss/rinex/obs/2021/001/brst0010.21d.Z
100% [.........................................................] 1943641 / 1943641
 File has been found  brst0010.21d.Z
No SNR file was created - check logs section for additional information
That took  3.69  seconds
Feedback written to subdirectory logs
(env) t@t-VirtualBox:~$ rinex2snr sc02 2008 1
Will seek the RINEX file from an external archive
sc02  year: 2008  doy: 1 from:  all
did not find lowercase station snr file
Requested a GPS only nav file
100% [.............................................................] 72961 / 72961found nav file at SOPAC
Orbit file:  /home/t/gnss/io/2008/nav/auto0010.08n
Searching the  unavco  archive for  sc02
https://data.unavco.org/archive/gnss/rinex/obs/2008/001/sc020010.08d.Z
100% [...........................................................] 609508 / 609508
 File has been found  sc020010.08d.Z

SUCCESS: SNR file was created: /home/t/gnss/io/2008/snr/sc02/sc020010.08.snr66
That took  9.25  seconds
Feedback written to subdirectory logs

the problem is with brst. it can work for sc02. These are what I type by using linux above

1026689430, 2023-03-07:
Broadcast:/home/t/gnss/io/2021/nav/auto0010.21n                                                                                               
 RINEX file:brst0010.21o                                                                                                                        
 Output file:brst0010.21.snr66                                                                                                                   
 Selection:66
 Decimation0            0
 ERROR:supports <=20 observ types                                                                                                    
 Coords   4231162.7879999997       -332746.91999999998        4745130.6890000002     
 message                                                                                                                             
 ERROR: this code only works for <= 20 obs types                                                                                     
 1 solution is to to run teqc on the original RINEX                                                                                  
 with -O.obs S1+S2+S5 as the option, rerun.      

This?

1026689430, 2023-03-07:
Receiver rate: low   
Decimation rate:   0 
Archive: all        
Orbits : nav        
The snrfile does not exist: /home/t/gnss/io/2021/snr/brst/brst0010.21.snr66    
SNR file brst0010.21.snr66                                  
 will use hybrid of python and fortran to make

kristinemlarson, 2023-03-07:
so this is telling you why it does not work.  there are too many observables in the file.

kristinemlarson, 2023-03-07:
For that particular site, you might as well use RINEX 3.  and specify the archive

rinex2snr brst00fra 2021 1 -archive sonel

it would also be better to use multi-GNSS orbits, 

rinex2snr brst00fra 2021 1 -archive sonel -orb gnss

1026689430, 2023-03-07:
it works,thank you so much,i'm new in this field.your help is very useful to me

closed 2023-03-13 by kristinemlarson