kristinemlarson/gnssrefl #89: why did not find the UNR database, I have station_pos.db file

1026689430, 2023-03-10:
Going to assume that you want to use the UNR database.
did not find the UNR database
Try to download the station database from github for you: https://github.com/kristinemlarson/gnssrefl/raw/master/gnssrefl/station_pos.db
If you are not online, it will fail.
Could not download the database for you
Tried to find coordinates in our UNR database. Not found so exiting

timdittmann, 2023-03-13:
@1026689430: can you provide some context for this issue, particularly with respect to how you are using the gnssrefl software.  The type of information we are looking for is covered in the following template: https://github.com/kristinemlarson/gnssrefl/blob/master/.github/ISSUE_TEMPLATE/bug_report.md

1026689430, 2023-03-28:
I downloaded, but it didn't work

timdittmann, 2023-04-03:
@1026689430 Thank you for the code:
Could you please add [additional context](https://github.com/kristinemlarson/gnssrefl/blob/master/.github/ISSUE_TEMPLATE/bug_report.md) so we can narrow our assistance?  Specifically aspects like OS, dockers, etc.  Thanks!

1026689430, 2023-04-05:
it is solved , thank you professor

closed 2023-04-05 by kristinemlarson