kristinemlarson/gnssrefl #174: Departure in selected points in final plot and in txt file

fenoglioluciana, 2023-08-09:
Screenshots If applicable, add screenshots to help explain your problem.

System (please complete the following information):

OS: [e.g. macOS, windows, etc...]

Method you are running the code:
Locally via github
Docker (please first make sure you are running the current docker image before reporting bug)
Jupyter notebook locally
Jupyter notebook via docker

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
1. commands run up to point of bug
2. command used when bug appeared
3. include error
4. Expected behavior A clear and concise description of what you expected to happen.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Method you are running the code**
* Locally via github
* Docker (please first make sure you are running the current docker image before reporting bug)
* Jupyter notebook locally
* Jupyter notebook via docker

**System (please complete the following information):**
 - OS: macOS, Windows, etc...

**Additional context**
Add any other context about the problem here.

fenoglioluciana, 2023-08-09:
Operation system Linux, the programs runs. We notice that when running the command 
daily_avg r5as 0.20 50 
we get a different number of points in the last figure generated and the in the txt file written in output.
We decided to modify the software generating a new file storing the points in the figure, this confirmed our first guess (different points). We would like to understand why is this done, which is the difference between the two sets of points and which data to compare with in-situ.
Many thanks for this clarification and for helping us to use the output of the software as intended. 
Best regards
Luciana

kristinemlarson, 2023-08-11:
Luciana, the daily_avg code creates more than one output.  When posing a question on github, I need you to write your question so that I know exactly which file you are asking about. I can't answer the question otherwise. It is also important so that others with the same question will be able to tell if this is relevant to issues they are also having.  Similarly, the figures all have names - so it would also be helpful if you would say which specific figure you are concerned about.  

I have used your question as an opportunity to look again at the code to see how I can make it clearer to use. I have rewritten the documentation, which is linked with other information here:

https://github.com/kristinemlarson/gnssrefl/issues/175

closed 2023-08-11 by kristinemlarson