kristinemlarson/gnssrefl #181: gnssir_input writes ediff as a string in the json file, resulting in an error when running gnssir

caradag, 2023-08-18:
OS: Linux / Ubuntu 20.04
Method you are running the code: Locally via github

**Describe the bug**
I was using an installation a few months old, and was facing this error using make_json_input. I was taking the workaround of just editing the jason file and removing the quotes. That is changing the line
"ediff": "30", 
by 
"ediff": 30,

Today I made an up-to-date installation, and I ran into the same problem using gnssir_input, so I'm filing this bug report.

**To Reproduce**
1. commands run up to point of bug
I was using data from a custom station named PAPO
run:
$ gnssir_input papo -lat 49.668203878 -lon -123.165468169 -height 135.273 -ediff 30

2. command used when bug appeared
$ gnssir papo 2023 223

3. include error
$ gnssir papo 2023 223 -doy_end 227
{'station': 'papo', 'lat': 49.668203878, 'lon': -123.165468169, 'ht': 135.273, 'minH': 0.5, 'maxH': 8.0, 'e1': 5.0, 'e2': 25.0, 'NReg': [0.5, 8.0], 'PkNoise': 2.8, 'polyV': 4, 'pele': [5, 30], 'ediff': '30', 'desiredP': 0.005, 'azval2': [0, 360], 'freqs': [1, 20, 5], 'reqAmp': [5.0, 5.0, 5.0], 'refraction': True, 'overwriteResults': True, 'seekRinex': False, 'wantCompression': False, 'plt_screen': False, 'onesat': None, 'screenstats': False, 'pltname': 'papo_lsp.png', 'delTmax': 75.0, 'gzip': False, 'ellist': []}
refraction file exists
requested frequencies  [1, 20, 5]
Using the New Way of Selecting Arcs
Augmented elevation angle list []
Requested azimuths:  [0, 360]
Using azimuths:  [0, 360]
Refraction parameters  998.8961260561704 10.5497571609598 1
Using:  /home/camilo/Programas/gnssrefl/02_Reflection_code_inputs/2023/snr/papo/papo2230.23.snr66
Results will be written to: /home/camilo/Programas/gnssrefl/02_Reflection_code_inputs/2023/results/papo//223.txt
Traceback (most recent call last):
  File "/home/camilo/Programas/gnssrefl/env/bin/gnssir", line 8, in <module>
    sys.exit(main())
  File "/home/camilo/Programas/gnssrefl/env/lib/python3.8/site-packages/gnssrefl/gnssir_cl.py", line 324, in main
    gnssir(**args)
  File "/home/camilo/Programas/gnssrefl/env/lib/python3.8/site-packages/gnssrefl/gnssir_cl.py", line 316, in gnssir
    guts2.gnssir_guts_v2(**args)
  File "/home/camilo/Programas/gnssrefl/env/lib/python3.8/site-packages/gnssrefl/gnssir_v2.py", line 208, in gnssir_guts_v2
    arclist = new_rise_set_again(thissat[:,1],thissat[:,2],thissat[:,3],e1, e2,ediff,satNu,screenstats)
  File "/home/camilo/Programas/gnssrefl/env/lib/python3.8/site-packages/gnssrefl/gnssir_v2.py", line 963, in new_rise_set_again
    **min_deg = (e2-ediff)   - (e1 + ediff)
TypeError: unsupported operand type(s) for -: 'float' and 'str'**

5. Expected behavior A clear and concise description of what you expected to happen.
That the ediff value in the resulting json file is a number not a string, this means that it has no quotes.
for ediff=4 the corresponding line in the json file should read:
"ediff": 4, 
instead of 
"ediff": "4",

kristinemlarson, 2023-08-18:
thank you for letting us know!  i have fixed it and am now testing the upload.  it will be version 1.5.5 hopefully in the not too distant future.

caradag, 2023-08-18:
Great! Thank you very much

closed 2023-08-18 by kristinemlarson