kristinemlarson/gnssrefl #196: Local Python Install: ISSUE

makanina, 2023-09-27:
Hi
I´m using the steps for Python installation by using Ubuntu, previously in my other Pc I used the version code from 2021 without issues, but now in other PCs I've had many problems installing it. 
On my last try, i activated the environments and run the commands: **pip install wheel** and **pip install gnssrefl** in the last one command i had this message: 
`
 ```
error: subprocess-exited-with-error

  × Building wheel for gnssrefl (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [403 lines of output]
      running bdist_wheel
      running build
      running config_cc
      INFO: unifing config_cc, config, build_clib, build_ext, build commands --compiler options
      running config_fc
      INFO: unifing config_fc, config, build_clib, build_ext, build commands --fcompiler options
      running build_src
      INFO: build_src
      INFO: building extension "gnssrefl.gpssnr" sources
      INFO: f2py options: ['--verbose']
      INFO:   adding 'build/src.linux-x86_64-3.8/build/src.linux-x86_64-3.8/gnssrefl/fortranobject.c' to sources.
      INFO:   adding 'build/src.linux-x86_64-3.8/build/src.linux-x86_64-3.8/gnssrefl' to include_dirs.
      INFO:   adding 'build/src.linux-x86_64-3.8/gnssrefl/gpssnr-f2pywrappers.f' to sources.
      INFO: building extension "gnssrefl.gnsssnr" sources
      INFO: f2py options: ['--verbose']
      INFO:   adding 'build/src.linux-x86_64-3.8/build/src.linux-x86_64-3.8/gnssrefl/fortranobject.c' to sources.
      INFO:   adding 'build/src.linux-x86_64-3.8/build/src.linux-x86_64-3.8/gnssrefl' to include_dirs.
      INFO:   adding 'build/src.linux-x86_64-3.8/gnssrefl/gnsssnr-f2pywrappers.f' to sources.
      INFO: building extension "gnssrefl.gnsssnrbigger" sources
      INFO: f2py options: ['--verbose']
      INFO:   adding 'build/src.linux-x86_64-3.8/build/src.linux-x86_64-3.8/gnssrefl/fortranobject.c' to sources.
      INFO:   adding 'build/src.linux-x86_64-3.8/build/src.linux-x86_64-3.8/gnssrefl' to include_dirs.
      INFO:   adding 'build/src.linux-x86_64-3.8/gnssrefl/gnsssnrbigger-f2pywrappers.f' to sources.
      INFO: building extension "gnssrefl.xnmeasnr" sources
      INFO: f2py options: ['--verbose']
      INFO:   adding 'build/src.linux-x86_64-3.8/build/src.linux-x86_64-3.8/gnssrefl/fortranobject.c' to sources.
      INFO:   adding 'build/src.linux-x86_64-3.8/build/src.linux-x86_64-3.8/gnssrefl' to include_dirs.
      INFO:   adding 'build/src.linux-x86_64-3.8/gnssrefl/xnmeasnr-f2pywrappers.f' to sources.
      INFO: build_src: building npy-pkg config files
      running build_py
      creating build/lib.linux-x86_64-3.8
      creating build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/check_rinex_file.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/computemp1mp2.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/daily_avg.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/daily_avg_cl.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/decipher_argt.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/download_ioc.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/download_noaa.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/download_orbits.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/download_psmsl.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/download_rinex.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/download_teqc.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/download_tides.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/download_unr.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/download_wsv.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/EGM96.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/filesizes.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/gnssir.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/gnssir_cl.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/gnssir_input.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/gnssir_v2.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/gps.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/gpsweek.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/highrate.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/installexe_cl.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/invsnr_cl.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/invsnr_input.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/karnak_libraries.py -> build/lib.linux-x86_64-3.8/gnssrefl
      copying gnssrefl/kelly.py -> build/lib.linux-x86_64-3.8/gnssrefl
[... 335 lines cut ...]
        Extra checks: none

      CPU dispatch  :
        Requested   : 'max -xop -fma4'
        Enabled     : SSSE3 SSE41 POPCNT SSE42 AVX F16C FMA3 AVX2 AVX512F AVX512CD AVX512_KNL AVX512_KNM AVX512_SKX AVX512_CLX AVX512_CNL AVX512_ICL
        Generated   : none
      INFO: CCompilerOpt.cache_flush[857] : write cache to path -> /tmp/pip-install-v3soyx9e/gnssrefl_56a46b941ada466e87385542848c5850/build/temp.linux-x86_64-3.8/ccompiler_opt_cache_ext.py
      error: Command "x86_64-linux-gnu-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Ibuild/src.linux-x86_64-3.8/build/src.linux-x86_64-3.8/gnssrefl -I/tmp/pip-build-env-dmyd3vnp/overlay/lib/python3.8/site-packages/numpy/core/include -Ibuild/src.linux-x86_64-3.8/numpy/distutils/include -I/home/meeb/GNSS/gnssrefl/env/include -I/usr/include/python3.8 -c build/src.linux-x86_64-3.8/gnssrefl/gpssnrmodule.c -o build/temp.linux-x86_64-3.8/build/src.linux-x86_64-3.8/gnssrefl/gpssnrmodule.o -MMD -MF build/temp.linux-x86_64-3.8/build/src.linux-x86_64-3.8/gnssrefl/gpssnrmodule.o.d -msse -msse2 -msse3" failed with exit status 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for gnssrefl
Failed to build gnssrefl
ERROR: Could not build wheels for gnssrefl, which is required to install pyproject.toml-based projects

[notice] A new release of pip is available: 23.0.1 -> 23.2.1
[notice] To update, run: pip install --upgrade pip
`

After that doesn't recognize the "installexe" command nor "installexe -h" command. 
 [image]

Some idea that i´ve been doing wrong ?

Many thanks for your time

kristinemlarson, 2023-09-27:
I have assigned your question to Tim. Please confirm that you followed these instructions before running pip install wheel,

[image]

makanina, 2023-09-27:
I confirm, I am following the instructions you mention

kristinemlarson, 2023-09-28:
One thing you can try is to install using pypi instead of git clone.
Instructions are on the same page, and I think you should use a different
folder and virtual environment.

kristinemlarson, 2023-09-28:
I believe you should also disclose which version of Ubuntu you are using.  I see from the logs you are using python 3.8

makanina, 2023-09-28:
Thanks you, i´ll try to install using pypi. 
And the Ubuntu version is 22.04.2

Many thanks

timdittmann, 2023-10-02:
Hi @makanina, 
Did you find success with the pypi install?
If not, can you share which versions of gcc/gfortran is trying to compile the package?
[If it helps, the outputs of `gcc--version` and `gfortran --version`]

kristinemlarson, 2023-10-05:
Did you get it things to work?  If there is a way to improve our documentation, please do let us know.

closed 2023-10-05 by kristinemlarson

makanina, 2023-10-05:
Thanks for your answers. 
I couldn't install it on my computer, but a colleague did manage to install it on his computer.
So I'm working on that computer.

Baizitang, 2023-12-21:
> 感谢您的回答。我无法在我的计算机上安装它，但一位同事确实设法将其安装在他的计算机上。所以我在那台电脑上工作。

Can you tell me the ubuntu version, gcc version and gfotran version of the computer you are working on?