kristinemlarson/gnssrefl #197: The "vwc" command is not working properly

Septyaz, 2023-10-01:
Screenshots If applicable, add screenshots to help explain your problem.

System (please complete the following information):

OS: Linux (Ubuntu)

**Describe the bug**
The "vwc" command for measuring soil moisture does not produce a .txt file, even though I have followed the command in the example case in the following link: https://gnssrefl.readthedocs.io/en/latest/pages/first_drivethru.html 

**To Reproduce**
I have tried all the sample cases provided on the following link https://gnssrefl.readthedocs.io/en/latest/pages/first_drivethru.html and followed the given commands. However, none of the sample cases produced a .txt file as described in the statement "These soil moisture estimates are written to a file; its location is written to the screen."

**Expected behavior**
According to the explanation provided on the link https://gnssrefl.readthedocs.io/en/latest/pages/first_drivethru.html in the case study of Boulder, Colorado, USA, after the "vwc" command such as "vwc mfle 2016 -year_end 2019," there should be a directory containing soil moisture estimate files in .txt format.

**Additional context**
If it's in accordance with the explanation, there should appear soil moisture estimate files in .txt format, as seen in the "Screenshot from 2023-10-01 22-29-46" image.

**Screenshots**
The result I obtained from the "vwc" command
[image]
[image]
[image]
[image]

The display after I ran the command.
[image]
[image]

The command and explanation that I followed.
[image]
[image]

An example of the soil moisture estimates file in .txt format that I expected.
[image]

kristinemlarson, 2023-10-02:
I habe an idea of why the file may not
Be created. I will try to test it today.

kristinemlarson, 2023-10-05:
My idea about why this was not working for you does not appear to be correct. I thought perhaps the file directory was not being created, and thus the file itself should not be created either.  
But I deleted my files and tried again and the directory is being created.  for the station p475, these are the last 
three lines that are printed to the screen.  So when I look, the file is in $REFL_CODE/Files/p475/p475_vwc.txt

Saving to /Users/kristine/Documents/Research/Files/p475/p475_phase_vwc_result.png
Saving to /Users/kristine/Documents/Research/Files/p475/p475_vol_soil_moisture.png
>>> VWC results being written to  /Users/kristine/Documents/Research/Files/p475/p475_vwc.txt

Does this file get created for you?  In looking at the results for p475, I don't like that there is a big unexplained gap in 2020. There is a reason for that gap, so I will be making some edits.  But for the issue of where the results go, I wanted to check back that you are (or are not) getting this filename printed to the screen and that it is being created.

The output for vwc used to go to $REFL_CODE/Files, but it was getting messy with results for so many sites.
SO I started putting things in a subdirectory by default

Kristine

kristinemlarson, 2023-10-09:
I would like to fix this for you - please let me know the current state of things when you have a chance.  As a separate manner, I have decided that p475 is not a very good example site.  So i have removed it from the example list for now. I still want to make sure the VWC code works - so if it is not creating output files in other test cases, I do want to know about that.

closed 2023-10-09 by kristinemlarson