kristinemlarson/gnssrefl #201: not allowed to add importlib, docker fails to build

kristinemlarson, 2023-10-26:
Tim, I tried to add importlib so I could reference version number in the output of gnssrefl.

It works for me locally.  

It failed at the docker creation end. 
(https://github.com/kristinemlarson/gnssrefl/actions/runs/6652947113/job/18077911919)

Preparing metadata (setup.py): started
#22 9.129   Preparing metadata (setup.py): finished with status 'error'

Do you know what is going on?

Kristine

kristinemlarson, 2023-10-26:
maybe it wants me to install it in the Dockerfile.  Anyway, you are the expert, not me.

kristinemlarson, 2023-10-28:
works now.

closed 2023-10-28 by kristinemlarson

timdittmann, 2023-10-30:
@kristinemlarson glad to hear you resolved this.  Let me know if you have any further questions on this one .