kristinemlarson/gnssrefl #205: I'm having some issues with rinex2snr

Baizitang, 2023-11-01:
Firstly, I used rinex2snr to process the local hkqt data in rinex3 format with four systems. When the file was named rinex2, the normal processing was completed

The command is:

`rinex2snr hkqt 2017 232 -doy_end 237 -snr 88 -orb gnss -strip T -screenstats T -nolook T`

Then, I used rinex2snr to extract the snr data of QDKZ of my own site, which also had four systems and was in rinex3 format, but when processing, it was recognized as rinex2.1, and the generated snr seemed to have data disorder.

I used gfz tool to convert to rinex2 version of the file, extract snr still wrong.

This is the snr code of 106 extracted from qdkz in rinex2.11 format

```
D:\Docker&gt; docker run -it --rm -v .\gnssrefl:/etc/gnssrefl/ gnssrefl bash

root@c9cfe1c243b2:/usr/src/gnssrefl# rinex2snr qdkz 2023 106 -snr 88 -orb nav -nolook T -screenstats T -strip T

Will first assume RINEX file  qdkz  year: 2023  doy: 106 is in the local directory

Found the RINEX 2.11 file qdkz1060.23o

Testing out stripping the RINEX 2 file here

You chose the strip option to reduce the number of observables in the RINEX file

Requested a GPS only nav file

Nav file exists online

Orbit file: / etc/gnssrefl/orbits / 2023 / nav/auto1060.23 n

SUCCESS: SNR file was created

/ etc/gnssrefl refl_code / 2023 / SNR/QDKZ/qdkz1060.23 snr88

That took 9.0 seconds

Feedback written to subdirectory logs

```

This is the snr code of 106 extracted from qdkz in rinex3 format

```
D:\Docker>docker run -it --rm -v .\gnssrefl:/etc/gnssrefl/ gnssrefl bash
root@6882e9810b48:/usr/src/gnssrefl#  rinex2snr qdkz 2023 106 -snr 88 -orb nav -nolook T -screenstats T -strip T
Will first assume RINEX file  qdkz  year: 2023  doy: 106 is in the local directory
Found the RINEX 2.11 file qdkz1060.23o
Testing out stripping the RINEX 2 file here
You chose the strip option to reduce the number of observables in the RINEX file
Requested a GPS only nav file
Nav file exists online
Orbit file:  /etc/gnssrefl/orbits/2023/nav/auto1060.23n

SUCCESS: SNR file was created
 /etc/gnssrefl/refl_code/2023/snr/qdkz/qdkz1060.23.snr88
That took  7.93  seconds
Feedback written to subdirectory logs
root@6882e9810b48:/usr/src/gnssrefl#

```

If you have time, could you help me see how to solve this problem?

The following screenshot shows the header of hkqt232.17o in rinex3 format
[image]

The following screenshot shows the header of qdkz1060.23o file in rinex2.11 format
[image]

The following screenshot shows the header of qdkz1060.23o file in rinex3.05 format
[image]

This is a screenshot of the file extracted by hkqt to snr88
[image]

This is a screenshot of the file extracted from qdkz of rinex2.11 to snr88
[image]

This is a screenshot of the file extracted from qdkz of rinex3 to snr88
[image]

Finally, this is the SNR extracted by a classmate using gnssSNR.exe (maybe my computer environment is a problem, I can't use this program), after I check with quickLook, I can get a lot of results as it should be, this is a screenshot of snr (99 format)
[image]

Thank you very much for taking time out of your busy schedule to check my question

kristinemlarson, 2023-11-02:
I am on travel right now - but a couple thoughts. First, you are letting the software use GPS orbits and not GNSS orbits. So I strongly urge that you say -orb gnss. Because otherwise there is really no reason to have RINEX 3 files at all.

As for RINEX 3, it is 100% required that you use a RINEX 3 station name (9 characters).  I think you are assuming you can still call it by the RINEX 2 name (which is four characters).  The RINEX 3 names are very complicated.  I would recommend you read here:

https://gnssrefl.readthedocs.io/en/latest/pages/file_structure.html

or in the rinex2snr documentation:

https://gnssrefl.readthedocs.io/en/latest/api/gnssrefl.rinex2snr_cl.html

For what it's worth, we allow RINEX 3 inputs, but the first thing it does is translate it to RINEX 2.11. The advantage
of letting gnssrefl do this conversion is that it will use the correct observables for GNSS-IR.  I am not convinced
that the defaults will do this for you if you try to convert yourself. At least this is what I saw when someone else i know tried to do the conversion himself.   

If I have misread your issue, I apologize. But as I said, I am on travel this week and cannot look more closely until I am back to my home base.

Kristine

Baizitang, 2023-11-02:
Thank you very much for taking time out of your busy schedule to answer my question. I'm sorry to disturb your trip. I sincerely wish you a pleasant trip.

I'll keep trying as you suggest, try rinex3 naming, try -orb gnss, and check again if there were any errors during the rinex2 conversion.

Thanks again for your reply

closed 2023-11-05 by Baizitang