kristinemlarson/gnssrefl #206: Invsnr "no reflector height data - exit"

christopherbarbarie, 2023-11-03:
Hi,

I'm having trouble running the invsnr code properly. I have gotten all my data from nmea into snr format and have been able to successfully run quicklook, gnssir, dailyavg and subdaily without a problem. I am able to run the invsnr_input code without a problem and it generates a .txt file however, when I go to run the invsnr code it exits with the following message: 

Found 0 arcs
no reflector height data - exit

Here is a screenshot of my inputs:

[image]
[image]

I'd be happy to provide more details if needed, any help would be appreciated thank you!

kristinemlarson, 2023-11-03:
hi - i think you have stumbled upon a bug that cropped up
when the calculation of the peak 2 noise ratio was changed.

two possible ways forward (if i am right)

install the latest version (1.9.1) and see if that fixes it

or

try when running invsnr -pktnlim 0 or some similar small value to see if
you can get some useful arcs.
the problem is that when the calculation was changed, the default
ratio was not changed.  at least that is how i recall it.

https://github.com/kristinemlarson/gnssrefl/issues/203

k.

kristinemlarson, 2023-11-03:
ps let me know if this is the solution, so i can close the issue.  or you can also close the issue if you like.

christopherbarbarie, 2023-11-03:
Hi, thank you for the reply, unfortunately upgrading to 1.9.1 did not fix the problem and neither did setting -pktnlim to 0 or any other small value.

kristinemlarson, 2023-11-04:
another idea

i looked at the code (it was written by david purnell).  he does not use
-100 to 50 as the rest of the code
would use for a northern region.  he would say 260 to 50 it seems.  so you
could try that, but it should have still looked for the
arcs from 0-50 so i'm not confident it will work.

the other thing i suggest is you say -screenstats T
at least that is the way i was able to see whether the northwest region was
being found or not.

so if using 260 to 50 does not work for you, i am willing to look at a
single snr file for you.  please gzip one and
send via google docs or your file sharing device of choice.  please do not
upload a snr file to github.
also send me your json so i can work out why invsnr is not producing
results.  it should be possible for me
to change david's code to allow both 260 to 50 and -100 to 50.

k.

christopherbarbarie, 2023-11-04:
Hi, unfortunately changing the azimuth limits did not work either, I attached a link to a dropbox folder with both an snr file and the json file. Thank you!

https://www.dropbox.com/scl/fo/3j6cfjah8rba3asc73qad/h?rlkey=n38w07t18mdxvnbpoe8gdf2ja&dl=0

kristinemlarson, 2023-11-05:
hi christopher

your SNR files are corrupt.  originally i thought you were storing two days
of data in one file and the SNR files can only have the data
for a single day in them as the time values are 0-86400.   but when i
looked closer i saw that you have duplicate measurements.
for example, if you only look at PRN 2 data (i.e. where column 1 is a 2),
and use a simple tool like grep,
grep 77015 prn2
  2   27.1159  310.6976   77015.0  0.000000   0.00  50.00   0.00   0.00
  2   27.1159  310.6976   77015.0  0.000000   0.00  50.00   0.00   0.00

you have two identical records. and that is not good.  i am not sure why
gnssir is so forgiving. so i guess the question is,
why two sets of identical measurements? is it a bug in nmea2snr or is it
that your NMEA file has duplicates?   so let me know
if you have an explanation on your end for the duplicates.

by the way, you should do the azimuths like this:

    "azilims": [
        260.0,
        50.0
    ],

because that is how david coded it up.  eventually i will let it take
negative azimuths, or at a minimum, document
this non-standard feature.

k.

christopherbarbarie, 2023-11-05:
Hi, thank you that fixed my problem. I had issues with the raw nmea files and onces I fixed that and reran nmea2snr everything worked fine.

Thanks again!

closed 2023-11-05 by christopherbarbarie

kristinemlarson, 2023-11-05:
glad it is working.  looked like you have very nice retrievals to the north,

k.