kristinemlarson/gnssrefl #224: confused by the documentation for docker install

kristinemlarson, 2023-11-18:
_If you have a lot of RINEX files and want to keep them organized, you should copy them into refl_code/rinex/station/yyyy/, where station is the lowercase 4char ID and yyyy is the year. You should then mount that directory in the docker run command as follows:_

Tim, I don't understand this documentation.  If the file is not in the local directory - the code expects the files to be in the directory called $REFL_CODE/yyyy/rinex/station.  

If you could update this documentation, that would be helpful. It would also be helpful to explain to people how to make sure their rinex file is in the local directory.

timdittmann, 2023-11-27:
- [ ] @timdittmann review docker instructions for mounting rinex file directory
- [ ] @timdittmann review docker instructions for where plots are saved locally
- [ ] @k-enloe and @timdittmann review instructions for jupyter install mounting directory of local rinex files

kristinemlarson, 2023-11-27:
@timdittmann - i am not questioning whether you added the right verbiage on how to correctly mount a directory for rinex files. my issue is that when people run the code and used your instructions, they will find that the code does not look for files in the directory.  that is what alerted me to the issue - when someone tried to use your directions.  

the issue is similar for @k-enloe. and it may not be relevant for the notebooks as i don't know if we have a notebook to help people that want to use their own data.  if that is the case, then perhaps the best response is to remove that sentence about mounting that directory entirely.  the refl_zones is probably a typo - but i did not want to change it without your say so.

for both of you, if you want to give people instructions on how to mount a rinex directory, you have to also tell them how to get the code to find it.

kristinemlarson, 2023-12-07:
I have done all I can do about this on my end, so I am closing this issue.  I will accept your PR when you make changes to the installs as we talked about.

closed 2023-12-07 by kristinemlarson