kristinemlarson/gnssrefl #234: docker builds are failing because the hatanaka download or install is failing

kristinemlarson, 2024-01-23:
tim - please look into this-  or remove the dependency on the GSI website.

kristinemlarson, 2024-01-24:
seems like an easy fix.  use the source we already have in gnssrefl.  

root@476f60f8cfb9:/tmp# wget https://terras.gsi.go.jp/ja/crx2rnx/RNXCMP_4.1.0_src.tar.gz
--2024-01-24 00:01:31--  https://terras.gsi.go.jp/ja/crx2rnx/RNXCMP_4.1.0_src.tar.gz
Resolving terras.gsi.go.jp (terras.gsi.go.jp)... 163.42.5.1
Connecting to terras.gsi.go.jp (terras.gsi.go.jp)|163.42.5.1|:443... connected.
ERROR: The certificate of ‘terras.gsi.go.jp’ is not trusted.
ERROR: The certificate of ‘terras.gsi.go.jp’ doesn't have a known issuer.

timdittmann, 2024-01-25:
Hi @kristinemlarson I am traveling but will make this change over the weekend.

kristinemlarson, 2024-01-25:
i uploaded the source you need to the proper source area.  it was living in docs, which does not help you.  
gnssrefl/crx2rnx.c
gnssrefl/rnx2crx.c

i think i did that right anyway.

timdittmann, 2024-01-29:
Addressed in [this PR](https://github.com/kristinemlarson/gnssrefl/pull/235).

kristinemlarson, 2024-01-29:
thank you tim

closed 2024-01-29 by kristinemlarson