kristinemlarson/gnssrefl #247: docker issues

kristinemlarson, 2024-02-23:
I am trying to provide a clean version of the code - and noticed the docker is not copying the files it is supposed to copy. Nor is it properly making directories.  

https://github.com/kristinemlarson/gnssrefl/actions/runs/8018033980/job/21903096249
line 938 and beyond says it is creating directories and so on.  And later says it is successfully copying files into them.  But they aren't there when the user then tries to use the docker.  

ls /etc/gnssrefl/refl_code

produces no files

while 

ls /etc/gnssrefl/exe
CRX2RNX  RNX2CRX  crx2rnx.c  gfzrnx  gfzrnx_2.0-8219_armlx64  gfzrnx_2.0-8219_lx64  rnx2crx.c

does.  it also creates files for kelly's work.

the docker says it is making at least the Files subdirectory below.(for reasons i do not understand doesn't make the input subdirectory until later). But it doesn't really matter because neither are there when you run the docker. at least not on my mac.

## executables
RUN mkdir -p /etc/gnssrefl/exe /etc/gnssrefl/orbits /etc/gnssrefl/refl_code/Files /etc/gnssrefl/notebooks

kristinemlarson, 2024-02-23:
FWIW, I  am following the instructions that we posted online for running the docker.

kristinemlarson, 2024-02-23:
I honestly don't know what you did when you tested this.  Maybe it worked before.  Maybe not.  I cannot tell.

kristinemlarson, 2024-02-23:
more of a question - it looks to me that like unlike the way you have saved snr files and outputs created using the docker, you have not done that for orbits.  i don't understand that.  orbits can be used over and over again and there is no reason to download them again.  maybe it has to do with how people will use the code?  it could definitely be they would rather download them over and over again.  so just confirming that this is the way the Dockerfile was set up.

timdittmann, 2024-02-26:
Hi @kristinemlarson, @k-enloe is out on vacation, I am looking into this.

timdittmann, 2024-02-26:
The volume mount is overwriting everything in `refl_code` in the container when the user first issues the run command from a specific local path.  
> I can certainly see that we do not have to create and move those files - as they are in the gnssrefl directory, but it seems odd to have a Dockerfile indicate that it is doing it. I think that would almost always confuse people.

I agree with this, and submitted [PR](https://github.com/kristinemlarson/gnssrefl/pull/248).

A summary of the logic if we revisit this ever and for @kristinemlarson review:
Inside the docker container, various input files (EGM96geoidDATA.mat, station_pos.db, gpt_1wA.pickle ) are shipped with the repository (e.g. /usr/src/gnssrefl/gnssrefl/gpt_1wA.pickle) that is copied into the container.  
When looking for these files inside the container for the first time from a specific local path, gnssrefl will copy these files from the repo path`/gnssrefl/` to `$REFL_CODE` ([e.g.](https://github.com/kristinemlarson/gnssrefl/blob/ab208adabb65082263c0b3c2c799ca47c04a5b33/gnssrefl/gps.py#L6664)), which inside the container is `/etc/gnssrefl/refl_code/`. This directory is in the volume mount and will persist locally for future requests.  😅

kristinemlarson, 2024-03-10:
tim and i discussed this - and i think the way forward is to better document the Dockerfile so we understand why it is creating those directories.  By the way, the station_pos.db has been updated.  it now has a 2024 in it - the code should use either, but the 2024 version has many more stations in it.

closed 2024-03-10 by kristinemlarson