kristinemlarson/gnssrefl #253: NMEA lat, long, height

jonauq2508, 2024-03-10:
[image]
Hi, I have been trying to analize NMEA format data, but I have a question.
Do I have to add the lat long and height manually? and if yes, how is this made? thanks in advance

kristinemlarson, 2024-03-11:
please check the online documentation for examples of nmea2snr
they aren't always there - but they are the best place
to answer questions. the examples are at the bottom after
the description of the purpose and inputs of the python function itself.

https://gnssrefl.readthedocs.io/en/latest/api/gnssrefl.nmea2snr_cl.html

other function links are listed here

https://gnssrefl.readthedocs.io/en/latest/pages/quick_recall.html#

kristinemlarson, 2024-03-11:
thanks for providing the screen shot.  that makes it easier for me. :-)

closed 2024-03-11 by kristinemlarson

jonauq2508, 2024-03-18:
thank you so much , I am currently runing the gnssrefl with success