kristinemlarson/gnssrefl #255: snr plotting demo

fgnievinski, 2024-03-11:
someone asked in the course how to plot raw SNR data

Kristine said the new generic plotting utility can be used.
https://gnssrefl.readthedocs.io/en/latest/api/gnssrefl.quickplt.html
https://github.com/kristinemlarson/gnssrefl/blob/master/gnssrefl/quickplt.py

it'd be nice to have a demo in the docs.

filing this here in case anyone wants to contribute with an easy task.

kristinemlarson, 2024-04-15:
the coding part of this is pretty much done.  as soon as i get a chance i will post it with version 3.1.5. what i will not do is make example plots, so if someone wants to do that, it would be great.  i am closing this issue now because i don't want someone to work on the coding part since it is done.

kristinemlarson, 2024-08-09:
as of version 3.6.6 there is some documentation about how to plot raw SNR data using quickplt.
https://gnssrefl.readthedocs.io/en/latest/pages/understand.html#looking-at-raw-snr-data

this is far from the original purpose of quickplt - but it will work for raw SNR data.  see the documentation for quickplt for other options.