kristinemlarson/gnssrefl #256: can you change gnssir_input to also store default options for daily_avg and subdaily?

kristinemlarson, 2024-03-13:
possible variables:

daily_avg would be median filter and required tracks

subdaily could be knots

if stored in the json, they would be used, but could be overriden by online parameters.

suggested by felipe nievinski

fgnievinski, 2024-03-16:
currently I'm using the following subdaily options:
-spline_outlier1
-spline_outlier2
-ampl
-sigma
-alt_sigma
-rhdot
-delta_out
-knots
-hires_figs
-subdir

kristinemlarson, 2024-03-20:
I will take care of this.

kristinemlarson, 2024-03-21:
this is easier to do when it is an optional parameter.  will need to think about how to do it for daily_avg where it is a required parameter.

kristinemlarson, 2024-03-21:
so in principle this is available for daily_avg now.  see the documentation for daily_avg for details.  going forward, these values will be written to the json by gnssir_input, but without chosen values.  you can continue to set them on the command line or you can set them to zero which triggers the code use the json values.  since they are required parameters, i could not see a way to change them to optional parameters which would trigger all sorts of problems for other users.  i will probably not be able to work on the subdaily parameters until next week.

kristinemlarson, 2024-03-22:
@fgnievinski if you set rhdot to False it won't create a spline output cause it never gets to the second part of the code.  is there any reason to have rhdot to False for you?  cause the default for rhdot is True and i don't want to add gnssir_input options that people don't need.

kristinemlarson, 2024-03-22:
most of this (i.e. subdaily) is now available.  it is being tested on github - and once it passes that it will be version 3.1.4
if things start crashing, people need to tell me.

closed 2024-03-22 by kristinemlarson