kristinemlarson/gnssrefl #261: Can't find where "logs/psfd_hybrid_error.txt" is located

jsppr1147, 2024-03-15:
Good day! I'm running the code on Docker Desktop on Windows 11. I was wondering if someone could help me with this issue regarding the rinex2snr process. 

I placed the RINEX file in the folder /refl_code/2023/rinex/psfd/ and ran the rinex2snr module

[image]

[image]

I kept on getting this error but I can't seem to find the psfd_hybrid_error.txt in my local directory to figure out what went wrong during the processing.

kristinemlarson, 2024-03-15:
try typing ls.
is there a directory called logs there?

kristinemlarson, 2024-03-15:
if that doesn't work, you could try this full pathname

/usr/src/gnssrefl/logs

closed 2024-03-15 by kristinemlarson