kristinemlarson/gnssrefl #264: NMEA 0183 FORMAT

jonauq2508, 2024-03-17:
Hi.
Thanks in advance for all your help
I have a question, I have a low cost NMEA receiver that collects the following data: 

$GNGGA,151518.210,0947.8145,N,08410.1147,W,1,07,1.29,1110.0,M,1.8,M,,*63
$GPGSV,3,1,12,09,60,260,33,04,49,178,25,08,46,024,36,02,43,134,28*73
$GPGSV,3,2,12,21,43,111,43,07,33,341,,17,20,220,,14,14,284,*72
$GPGSV,3,3,12,27,11,034,,03,10,175,,30,08,322,,22,04,267,*76
$GLGSV,3,1,10,81,52,071,38,82,48,158,,69,37,268,,68,30,205,*62
$GLGSV,3,2,10,79,23,004,32,78,16,057,35,83,13,188,,88,09,034,*65
$GLGSV,3,3,10,80,06,315,,70,01,320,*6A
$GNRMC,151518.210,A,0947.8145,N,08410.1147,W,5.82,124.75,170324,,,A*6D

My question is, are this NMEA sentences valid to perform the transformation to SNR ?

thanks, greetings from Costa Rica

kristinemlarson, 2024-03-17:
@MakanAKaregar can you answer this question please ? thanks, kristine

jonauq2508, 2024-03-17:
[image]
 thanks for your help, I am running the GNSSREFL and when I analize my own data I got some errors, like this:

[image]
 it creates the snr file but it is empty

jonauq2508, 2024-03-17:
I have been also trying with nmea data from the example of Felipe N. and Manuella F. (Using gnssreﬂ for processing data from a low-
cost GNSS-IR sensor at Guaíba Lake, Brazil) and with their data I can get good results. 
[image]
the data from their NMEA is a little bit different from mine

jonauq2508, 2024-03-18:
This is the nmea data that I can analize ( this data is from a low cost NMEA receiver from the Felipe and Manuella example)
[image]

kristinemlarson, 2024-03-18:
It is good news you can analyze the data from @fgnievinski.  I can't answer your question as @MakanAKaregar wrote nmea2snr and understands the NMEA format. I think he will want you to provide more information than the code created an empty file.  Like the output to the screen when it was running.  But maybe he will know the answer simply by looking at the NMEA file you posted.

MakanAKaregar, 2024-03-18:
@jonauq2508 Your NMEA sentences look fine to me. However, the current nmea2snr tool does not support the conversion of a single block. You need to provide around 5 minutes of data for this conversion. In the very near future, I will be adding this capability into nmea2sntr for real-time applications.

MakanAKaregar, 2024-03-18:
@jonauq2508 It doesn't find the orbit because the GFZ SP3 today's file is available tommorrow. If you insist on processing today's data, you have the option to use orbit information from NMEA data by specifying -risky True. The quality of this orbit may vary depending on your receiver. Again, make sure you have a minimum of 5 minutes of data for this conversion. The data you've attached has less than 1 min of data.

jonauq2508, 2024-03-18:
Thank you very much for your response @MakanAKaregar . I will follow your advise, 
Greetings from Costa Rica

fgnievinski, 2024-03-19:
> I have a question, I have a low cost NMEA receiver that collects the following data:
> ```
> $GNGGA,151518.210,0947.8145,N,08410.1147,W,1,07,1.29,1110.0,M,1.8,M,,63 [...]

these data contain multiple GNSS: GPS and GLONASS. the original code was GPS only. later, it was extended to support GLO, too, see #44.

kristinemlarson, 2024-03-19:
@MakanAKaregar @fgnievinski @jonauq2508 

please would someone download and check newest version of nmea2snr.  it should look for final, rapid, and ultrarapid orbits
in nmea2snr.  it is difficult for me to check any of this as i do not have access to relevant NMEA files.

k.

naoyakadota, 2024-03-19:
Hello,

I happened to have an NMEA file generated within the last 24 hours and the code worked without any issues. I appreciate your update!

[image]

kristinemlarson, 2024-03-20:
Thank you!  If the GFZ ultra rapid orbit turns out not to be super reliable, someone could add wum or wum2 as an orbit option. The latter was added @MeanLowDrew is relatively recently. It accesses an  ultra rapid orbit from Wuhan but from Wuhan servers.  Apparently there were some issues getting the Wuhan files from CDDIS.

jonauq2508, 2024-03-20:
> Hello,
> 
> I happened to have an NMEA file generated within the last 24 hours and the code worked without any issues. I appreciate your update!
> 
> ![Screenshot 2024-03-19 at 3 15 15 PM](https://private-user-images.githubusercontent.com/65903698/314199257-1ea71324-75e0-4bbf-9498-4923a6ae9542.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA5NDE5ODYsIm5iZiI6MTcxMDk0MTY4NiwicGF0aCI6Ii82NTkwMzY5OC8zMTQxOTkyNTctMWVhNzEzMjQtNzVlMC00YmJmLTk0OTgtNDkyM2E2YWU5NTQyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIwVDEzMzQ0NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI4OTgwZGJhZTNkMzgzNTQ1ZDViNzQ0OWRmN2Q5MGYzZjRjNWY4ZTkyNDYzZDcwMDBiNDA5OWZlYjM4MDkwY2EmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.U-igJkOJ5R2qrgmNCcLel849y7Na3U4wE_pcW486-oA)

Hi, I see that you could not get the orbtis

jonauq2508, 2024-03-20:
hi @kristinemlarson I have check it the the newest version of na2s and there is still a problem getting the orbits files from GFZ, my data is from yesterday

naoyakadota, 2024-03-22:
@jonauq2508 For further clarification, GFZ is providing three different types of sp3 orbits - final(typically published a week after the event), rapid(in a day), and ultra-rapid (within a few hours). The message on the screenshot I have attached to the previous communication suggests that the GFZ has not published the final GFZ orbit yet, which is correct. 
[image]
If you refer to the original nmea2snr.py code, you may find that if the code cannot find any of the three gfz-sp3 orbits you will get some more messages showing up on your screen. Hence, you could assume that the code has successfully downloaded the "rapid orbit" instead and that's why you don't see any further message. Hope that helps.

kristinemlarson, 2024-03-22:
@jonauq2508 and @naoyakadota 

it could also be that you are inadvertently running the old version (old docker) where it was still expecting the final GFZ orbits.  even if you have updated your docker, you might still find yourself using the wrong one (i have done this).  so all i can say is that naoya is correct.  if you run the new version of the code, which you can confirm by checking which version is running in your docker, this problem was fixed.

closed 2024-03-22 by kristinemlarson

jonauq2508, 2024-03-22:
@naoyakadota and @kristinemlarson  thank you very much for the clarification and the assitance