kristinemlarson/gnssrefl #265: CSVfile option not working

christopherbarbarie, 2024-03-18:
I'm trying to output the results of subdaily to a csv file instead of the nominal txt file. I've tried using -csvfile True and -csv True but nothing changes and I keep getting the output as txt files.

kristinemlarson, 2024-03-18:
i think i got tired of maintaining that option since there are so many
files created by that code.
i'll see if i can resurrect it for you.

kristinemlarson, 2024-03-18:
do you want the individual RH results or do you only care about the smooth
spline at the end?

kristinemlarson, 2024-03-18:
is like this ok?  i have not done the lomb scargle files - just this last one.

%  station sc02 gnssrefl v3.1.0
%  This is NOT observational data - be careful when interpreting it.
%  If the data are not well represented by the spline functions, you will
%  have a very poor representation of the data. I am also writing out station
%  orthometric height minus RH, where Hortho (m) is      5.801
%  MJD,             RH(m), YYYY, MM, DD, HH, MM, SS,  quasi-sea-level(m)
%  (1),              (2),  (3), (4),(5), (6),(7),(8),   (9)
  60310.0000000,      5.016,2024,  1,  1,  0,  0,  0,     0.785
  60310.0208333,      5.023,2024,  1,  1,  0, 30,  0,     0.778
  60310.0416667,      5.034,2024,  1,  1,  1,  0,  0,     0.766
  60310.0625000,      5.048,2024,  1,  1,  1, 30,  0,     0.752
  60310.0833333,      5.067,2024,  1,  1,  2,  0,  0,     0.734
  60310.1041667,      5.096,2024,  1,  1,  2, 30,  0,     0.705
  60310.1250000,      5.141,2024,  1,  1,  3,  0,  0,     0.660
  60310.1458333,      5.207,2024,  1,  1,  3, 30,  0,     0.594
  60310.1666667,      5.299,2024,  1,  1,  4,  0,  0,     0.502

kristinemlarson, 2024-03-18:
It is the spline fit after the RH dot correction.  so the very last plot.  i will go ahead and post this to github.

christopherbarbarie, 2024-03-18:
perfect thank you!

kristinemlarson, 2024-03-18:
github runs some smoke tests to make sure i didn't screw things up - and it also makes and tests the docker.  so in about 10 minutes it should be ready to use.  i have meant to send out an email today but have not gotten to it yet.  next time you use gnssir try -par 10. we are parallel processing now!  also beta testing it in rinex2snr as long as you don't use a bad archive.

christopherbarbarie, 2024-03-18:
The function saves the file as a csv but just a heads up the file "stationname_spline_out" still saves a .txt file but it is in csv format

kristinemlarson, 2024-03-19:
current version 3.1.1 makes a file with csv file ending.

when i get a chance i will turn the csv endings back on for the other files.

kristinemlarson, 2024-03-20:
i believe the csv option in subdaily has come back from the dead. the input is now -csv instead of -csvfile.
as of version 3.1.2

closed 2024-03-20 by kristinemlarson