kristinemlarson/gnssrefl #277: Difficulty Reading GPS L2/S2W Data with rinex2snr: Seeking Explanation for Missing Signal-to-Noise Ratios

Baizitang, 2024-05-15:
I noticed that when using rinex2snr, the GPS L2 or S2W data was not successfully read and written. I tried both Rinex 3.02 and Rinex 3.05 after performing format checks using gfzrnx, but encountered the same issue. However, what's strange is that last year, when I converted SNR files using the old gnssrefl, the L2 signal-to-noise ratios were included. For other satellite systems and bands, everything seems normal, although I haven't conducted an in-depth comparison. Could you please explain what might be causing this?

`
root@fb8bfc44be2c:/usr/src/gnssrefl# rinex2snr qdkz00xxx 2023 105 -snr 88 -orb gnss3 -rate high -nolook T -screenstats T
Only one day being analyzed, parallel processing turned off
No parallel processing
/etc/gnssrefl/refl_code/2023/snr/qdkz/qdkz1050.23.snr88
SNR file does not already exist. Which means I will try to make it.
/etc/gnssrefl/refl_code/2023/rinex/qdkz/
Will first assume RINEX file  qdkz  year: 2023  doy: 105 is located here : /usr/src/gnssrefl
try looking for RINEX 3 in  /etc/gnssrefl/refl_code/2023/rinex/qdkz/
Apparent Rinex version 3.02
The RINEX 3 file exists locally QDKZ00XXX_R_20231050000_01D_01S_MO.rnx
Minimal feedback is written to  /etc/gnssrefl/refl_code/logs/qdkz/2023//105_translation.txt.gen
GFZ0MGXRAP_20231050000_01D_05M_ORB.SP3 GBM0MGXRAP_20231050000_01D_05M_ORB.SP3
ftp://ftp.gfz-potsdam.de/pub/GNSS/products/mgex/2257_IGS20/gbm22576.sp3.Z
ftp://ftp.gfz-potsdam.de/pub/GNSS/products/mgex/2257_IGS20/GBM0MGXRAP_20231050000_01D_05M_ORB.SP3.gz
100% [..........................................................................] 1043915 / 1043915Orbit file:  /etc/gnssrefl/refl_code/2023/sp3/GBM0MGXRAP_20231050000_01D_05M_ORB.SP3
Seeking permission from Earthscope to use their archive - wish me luck
That download experience took  1  seconds.

SUCCESS: SNR file was created
 /etc/gnssrefl/refl_code/2023/snr/qdkz/qdkz1050.23.snr88
That took  152.46  seconds
`
[image]
[image]
[image]

kristinemlarson, 2024-05-15:
I don’t think I allow L2W SNR data.

I am away from my code for a few days,
So can look at it then.

Do you have access to proper L2C signals?
They are far superior to L2 W

I get the names mixed up but I think
I have that right.

Kristine M. Larson
***@***.***
https://www.kristinelarson.net <http://www.kristinelarson.net>
https://gnss-reflections.org
https://github.com/kristinemlarson

Are you waiting for your code to be perfect?

kristinemlarson, 2024-05-16:
I think I can add a flag to allow you to keep s2w but i will
Never make it the default as it is a
Very low quality data.

If you previously used rinex 2.11 it would not throw out
L2w as it is more complicated to know what is stored in
Rinex 2.11 s2. That format does not allow you to have both
Kinds of l2 data.

Let me know if you must have the l2w data and i will see if
I can add a flag. But not until friday.

Kristine M. Larson
***@***.***
https://www.kristinelarson.net <http://www.kristinelarson.net>
https://gnss-reflections.org
https://github.com/kristinemlarson

Are you waiting for your code to be perfect?

kristinemlarson, 2024-05-17:
do you want me to add a flag so you can keep L2W?

Baizitang, 2024-05-18:
I apologize for missing your message and not responding in a timely manner.

I understand that if L2W performs poorly, it would be reasonable to discard it. If L2C is accepted in the observation file, it would be placed in the third column of the SNR data, correct? Can I assume that if the current SNR data file includes L2C, it cannot also include L2W? What happens if both L2W and L2C are present in the observation file?

I believe that having an additional option could improve the code, but if L2W is indeed too unreliable, there is no need to waste your time on it.

Thank you for all your contributions to gnssrefl.

Best regards

kristinemlarson, 2024-05-18:
the SNR format used in my software (the snr66 files) does not have any
space in it for multiple L2 observables.

*The SNR data format* part of
https://gnssrefl.readthedocs.io/en/latest/pages/file_structure.html

tells you which observables are in which columns.

if your files are in RINEX 3 the code will never write out L2W.  it only
writes out L2C observables.

if your file is RINEX 2.11, the code does not try to figure out which is
L2W and which is not.
it writes out whatever is defined as S2.  So typically such a file would
have a mixture of
observations in the SNR file, some would be L2C and some would be L2W.  it
is not ideal.

kristine

Baizitang, 2024-05-23:
I think I've got it figured out. I'll rule out s2w. Thank you for your answer.

closed 2024-05-23 by Baizitang