kristinemlarson/gnssrefl #278: Issues regarding rinex2snr processing

jsppr1147, 2024-05-15:
I accidentally downloaded the wrong version of the Rinex file for my processing (ver 2.10) and got an error saying that I needed a 2.11 version. So, I deleted the older version and downloaded the correct version 2.11 
[image]

[image]

but I still keep getting the same error. 

[image]

Any ideas on how to solve this issue?

kristinemlarson, 2024-05-15:
Can you post the log? The location should be written to the screen.

If that doesn't work I would need you
To post the rinex file somewhere but
DO NOT post it on github.

I am away from my code for a few days
But can look at it on Friday.

Kristine M. Larson
***@***.***
https://www.kristinelarson.net <http://www.kristinelarson.net>
https://gnss-reflections.org
https://github.com/kristinemlarson

Are you waiting for your code to be perfect?

jsppr1147, 2024-05-15:
i think i found the location of the log, the issue I have now is I cant open the folder in the container. It closes in split second as I open the folder

https://github.com/kristinemlarson/gnssrefl/assets/159399535/59ee4b73-bed0-4379-8bc6-1c77acd18dac

kristinemlarson, 2024-05-16:
Go ahead and send me the location of your  rinex file.
I will test it when i am back home

It could be you dont have any non zero snr data.
Or that you dont have any data below thirty degrees elevation angle.
I am a little surprised you only have one hour of data.
Not sure how useful that will be.

Please confirm what version of the gnssrefl code you are running.

Kristine M. Larson
***@***.***
https://www.kristinelarson.net <http://www.kristinelarson.net>
https://gnss-reflections.org
https://github.com/kristinemlarson

Are you waiting for your code to be perfect?

kristinemlarson, 2024-05-17:
I am back with my laptop.  please send me the location of a rinex file to test.

kristine

Baizitang, 2024-05-18:
I noticed that your observation file does not have coordinates and shows as 0, 0, 0. This might be the reason for the issue. You can manually add the coordinates to the observation file and try again. I have encountered this problem before, and I hope this helps you.

> I accidentally downloaded the wrong version of the Rinex file for my processing (ver 2.10) and got an error saying that I needed a 2.11 version. So, I deleted the older version and downloaded the [...]

kristinemlarson, 2024-05-18:
i don't know where you are on this - but you do have to have station
coordinates in the rinex file.
and the screenshot does not have them.  (APPROX POSITION XYZ)

kristinemlarson, 2024-05-18:
APPROX POSITION XYZ cannot be zeroes.

closed 2024-05-19 by jsppr1147