kristinemlarson/gnssrefl #279: Unable to generate SNR

zihelyu, 2024-05-22:
The program I installed using docker cannot generate SNR after running it. Is there something wrong with the command I ran? Thank you for answering this for me.
[image]
[image]

kristinemlarson, 2024-05-22:
I don't know what the problem is.  If you can post the rinex file somewhere public (but not on github and maybe gzipped), I will look at it.

zihelyu, 2024-05-22:
> I don't know what the problem is. If you can post the rinex file somewhere public (but not on github and maybe gzipped), I will look at it.

Already uploaded to this link
[](http://kh.dawnwl.com/zjgh2370.20o)

kristinemlarson, 2024-05-22:
i know for sure that my code will not allow 30 observable types.  there is
no
reason to have doppler (all the observations starting with D).
i will try to change the code so that it gives you a better warning about
this.

in the mean time, try this command.

smoosh zjgh2370.20o 15

but run it where the file exists.  i.e. ls zjgh* should be where you run
the command.

this will remove all the extra observables and only keep SNR. and the 15
decimates so current
sampling, which is 15.    and then try rinex2snr.

kristinemlarson, 2024-05-22:
I investigated a bit further and there is something about your file that
does not
like the line where I tested for RINEX version number:

rinext =np.loadtxt(r,usecols=0,dtype='str',max_rows=1)

it works for RINEX files from typical GNSS archives.

I am not sure why it does this - and I could provide a flag as a workaround
- but I don't think that will
help you much because my code does not allow RINEX 2.11 files with more
than 25 observables.
So you will need to fix them regardless.  And since most of those
observables are zero, it is best
that you stop writing them out.

I will see if I can figure out why your file triggers this issue.  Thanks
for letting me know about it.

kristinemlarson, 2024-05-22:
from what I read you are creating RINEX files with incorrect strings, i.e.
they are using
Latin instead of whatever python expects (utf-8).  i think smoosh uses
gfzrnx and maybe
it is friendlier to different strings.

kristinemlarson, 2024-05-22:
yeap, if i change the loadtxt command to say encoding = 'latin-1', it does
not crash. i could put it in a try/except i guess....

zihelyu, 2024-05-22:
> (utf-8

I am running it for my friend. I don’t know anything about GPS. Can you give me the running command of rinex2snr corresponding to this file? Thank you very much for answering my questions.

kristinemlarson, 2024-05-22:
I have just pushed a new version of the gnssrefl code to github.  It will take about 15 minutes to make a new docker.  When that version is ready, please **update your docker**, and then test with your existing RINEX file that you were storing on your system  with this command:

rinex2snr zjgh 2020 237   -orb gnss -strip T -nolook T

the extra boolean command strip will remove those extra observables I was previously talking about.   and the command that was failing is now behind a try/except.

zihelyu, 2024-05-22:
> zjgh2370.20o

OK, I have successfully run and generated images, thank you very much for your help.

kristinemlarson, 2024-05-22:
i will make a new version number for this later today.  thanks again for helping me figure this out.

closed 2024-05-22 by kristinemlarson