kristinemlarson/gnssrefl #285: moving to new python versions

kristinemlarson, 2024-06-29:
we are currently limiting which python version we are using (<= 3.11 ) because 
of a library in numpy (distutils) needed to use f2py.  this eventually
has to be addressed.  if anyone has special skills to help make this conversion easier, it would be much appreciated.  looks here like the solution going forward is to use meson. 

https://numpy.org/doc/stable/f2py/buildtools/index.html

Baizitang, 2024-07-10:
Additionally, while upgrading and adapting, is it possible to configure the environment and complete the software operations locally on a Windows system? Given that a significant number of people use Windows, switching to a virtual machine running Linux or using WSL2 can be quite complex.

kristinemlarson, 2024-07-10:
Have you tried using the Docker?

Baizitang, 2024-07-11:
> 你试过使用Docker吗？
> […](#)
> -- Kristine M. Larson ***@***.*** https://www.kristinelarson.net <http://www.kristinelarson.net> https://gnss-reflections.org https://github.com/kristinemlarson Are you waiting for your code to be perfect?

Yes, I am using the Docker version, but I find it somewhat complicated to edit code within Docker. Perhaps I am not using the right method? Do you have any suggestions for this? Any assistance you could provide would be greatly appreciated.

kristinemlarson, 2024-07-11:
I have not myself modified code while running the Docker.  I would point you to @andriaNOAA as she has expressed an interest in a Windows version of gnssrefl in the past.  And I know one of her colleagues contacted me as well, though I can't remember his name now.  I am not sure what the current status of gnssrefl usage at NOAA.

If we switch to using plain fortran instead of the fortran within python, and the Windows users have a fortran compiler pre-installed, I imagine it might be possible to get a Windows version running.

kristinemlarson, 2024-07-15:
or perhaps @timdittmann could comment on modifying code when using the docker.  he is the docker expert.

Baizitang, 2024-07-16:
> or perhaps @timdittmann could comment on modifying code when using the docker. he is the docker expert.

I am currently exploring the use of VS Code and its plugins to edit Docker containers under WSL2 on Windows. I greatly appreciate your suggestions, and if the opportunity arises, I will certainly consider seeking their assistance as well. Thank you very much for your support.

timdittmann, 2024-07-16:
> I am currently exploring the use of VS Code and its plugins to edit Docker containers under WSL2 on Windows. I greatly appreciate your suggestions, and if the opportunity arises, I will certainly [...]

@Baizitang I agree that [VS Code for your docker dev environment](https://code.visualstudio.com/docs/devcontainers/containers) sounds like a good strategy.  
I cannot speak from experience to the interaction with the WSL/Windows, but I do know VSC is developed by Microsoft. 🤞 

Maybe someone in this opensource world will have other suggestions?