kristinemlarson/gnssrefl #301: Can subdaily use multiple azimuth constraints?

jsppr1147, 2024-07-28:
Our area has only vision for 0-60, 100-160, and 300-360 azimuth for water reflections. I'd like to know if there is any workaround for this setup, instead of doing it by parts.

kristinemlarson, 2024-07-28:
i'm not sure what you mean.  you would ordinarily put those azimuth limits
in the json (in gnssir_input)
and it would save them for you.  and when you run gnssir,
the results file would have all the region reflector height results mixed
up together
and sorted by time.  so i'm not sure what you mean by parts?

gnssir_input station -azlist2 0 60 100 160 300 360

the latest version of the code has a local coordinates file so i urge you
to use it (llh_local.txt).  that way you would not
have to input the lat lon and height on the command line every time using
gnssir_input.  the option is described
in the file formats description page

https://gnssrefl.readthedocs.io/en/latest/pages/file_structure.html

kristinemlarson, 2024-07-28:
so now i see that you are asking about subdaily.  i still don't understand
the question - if you use the azlist2 option in gnssir, subdaily will
be able to pick up all the results together.  is the problem that you are
using a very old version of the software, and thus had
set up different jsons for different azimuths?  which version are you using?

jsppr1147, 2024-07-28:
Ohh. I see. I got confused on the -azi1 -azi2 on subdaily. i thought it uses different azimuth constraints aside from the gnssir_input. thanks for the clarification :)

closed 2024-07-28 by jsppr1147