kristinemlarson/gnssrefl #313: NameError: name 'gnssrefl_helpers' is not defined

HadiShali, 2024-09-05:
Hi, 
I'm trying to run the docker version of gnssrefl however, I'm having issues with importing the library of the gnssir package. I have attached a screenshot which you can see where the problem is. I would be appreciate it if you could check it and help me out. 

Thanks,
[image]

kristinemlarson, 2024-09-05:
Kelly, this is the same issue as before. pls fix this, Thank you.

Kristine M. Larson
***@***.***
https://www.kristinelarson.net <http://www.kristinelarson.net>
https://gnss-reflections.org
https://github.com/kristinemlarson

Have you tried reading the documentation?

kristinemlarson, 2024-10-06:
I asked Earthscope - and this is what they said.  

-------------------------------------------------------------------------------------------------------------------------

Thank you for writing to us about this issue. We will have someone look into the listed issue on GitHub soon. If it’s a small, easily addressed problem, we should be able to see this resolved in short order.

In general, we place maintenance of notebooks (and many of our apps) that are not in active production on the Low Priority list and I think it would be really healthy for our community to be willing to apply their expertise to provide fixes and enhancements to the notebook collection that can be reviewed in a pull request and merged for deployment.

We’ll keep this ticket active in the short term and write back on what the resolution outlook appears to be. Thank you!

kristinemlarson, 2024-10-08:
I am officially closing this issue as it is not related to the code itself, but rather specifically to the notebooks. It is still possible someone at Earthscope will update those notebooks, and I will be happy to accept the PR if it is submitted.

closed 2024-10-08 by kristinemlarson