kristinemlarson/gnssrefl #314: rinex2snr seems to search for RINEX files online even when using local RINEX files with the -nolook T command

Baizitang, 2024-09-11:
Hi,

I've encountered an issue while using the tool. When I use local RINEX files and the command, it still appears to search for RINEX files online. I am unsure if this is a bug in the program or just an incorrect output message.rinex2snr-nolook T

To provide more context, I will demonstrate two scenarios over two days:

In most cases, when it doesn't find the file I've locally named, it skips the search and runs normally.
However, in one case, it seems to find and search for RINEX files online, which should not happen with .-nolook T
Any insights on whether this behavior is intentional or an error would be helpful.

Thank you!!
1.code
[image]
2.doy118
[image]

3.doy119 （I have omitted some repetitive content in the middle.）
[image]
[image]

kristinemlarson, 2024-09-12:
please try the new version if you can.  it should no longer look for files when it already has one!

Baizitang, 2024-09-13:
Thank you very much for your efforts, after I tried this issue I think this issue has been solved, but I have a new problem and don't know if I should continue to give feedback here.

I used the -gnss3 command to do this, but when downloading the track, the regular download part stopped, what is this error?

Should I download and upload the sp3 to the folder and solve it, and if so, how do I command to identify my local sp3?

1. code
[image]

1. The run was successful
[image]

3. sp3 (GFZ FIN) download issue
[image]

kristinemlarson, 2024-09-13:
I am away from my laptop and cannot
Help you with this now.

I would recommend using another orbit
Source.

Kristine M. Larson
***@***.***
https://www.kristinelarson.net <http://www.kristinelarson.net>
https://gnss-reflections.org
https://github.com/kristinemlarson

Have you tried reading the documentation?

Baizitang, 2024-09-13:
Yes, I have read the documentation, as I used the same orbit for my previous data, and my goal is to process it with the same orbit. Thank you, I think I will try some other tracks instead.

Baizitang, 2024-09-13:
Ok，thank you，I will try it.

Baizitang, 2024-09-17:
OK，I get it！ Thanks.

closed 2024-09-17 by Baizitang