kristinemlarson/gnssrefl #318: ModuleNotFoundError: No module named 'gnssrefl.gpssnr'

cheesepalm, 2024-10-12:
I used git clone and then pip install
'pip install .'

Working on Linux

I installed the package and it said "Successfully built gnssrefl"
Then I tried to open rinex2snr -h
and it resulted in
ModuleNotFoundError: No module named 'gnssrefl.gpssnr'

I use a virtual environment with python 3.9.20.

pip list | grep gnssrefl
gnssrefl                  3.10.0

Can you help me with this? I am a bit puzzled at the moment.

cheesepalm, 2024-10-12:
Update: Okay I'm not 100% sure what I did but I think that using pip3 instead of pip did the trick for me. Now the rinex2snr program can be started normally.

closed 2024-10-12 by cheesepalm

kristinemlarson, 2024-10-12:
Thank you for posting this. I will make a note in the install instructions
to warn people that pip3 may be needed.

Kristine M. Larson
***@***.***
https://www.kristinelarson.net <http://www.kristinelarson.net>
https://gnss-reflections.org
https://github.com/kristinemlarson

Have you tried reading the documentation?

dahmic77, 2024-10-29:
Hi, I have the same issue. I followed your instruction under "Direct Python Install" (https://gnssrefl.readthedocs.io/en/latest/pages/README_install.html):
`pip install . `
`installexe linux64  (Note: The EXE environment variable has to be set via .bashrc and also the EXE directory has to be created before this command work.)

Version gnssrefl-3.10.2 was installed at my Linux PC. Then I tried to run 'rinex2snr -h' command and got the same failure:
**ModuleNotFoundError: No module named 'gnssrefl.gpssnr'**

'gpssnr.f' is Fortran code. In a former version of gnssrefl a 'build' directory was created with Fortran binaries. But it seems to be that the installation steps above do not include a compilation of the Fortran code at the moment. Is the compilation of the Fortran code not necessary anymore?

kristinemlarson, 2024-10-29:
hi are you using the docker?  a linux emulator on your PC?  which python
are you running?  i can't say much without
more information and a screen shot.

kristinemlarson, 2024-10-29:
and are you using python3 and pip3 (or pip pointing to python3)

dahmic77, 2024-10-29:
I solved the problem. See solution under https://github.com/kristinemlarson/gnssrefl/issues/320. The 'rinex2snr -h', 'gnssir -h' or 'subdaily -h' commands should not be used in the ./gnssrefl installation directory itself. Otherwise it works.

kristinemlarson, 2024-10-29:
i can run those same commands anywhere.  so i'm not sure what you are
saying.

dahmic77, 2024-10-29:
I can also run the command anywhere except the under the main ./gnssrefl directory. I do not know, what is the reason for that. 
[image]

kristinemlarson, 2024-10-29:
can you get the code to do anything useful? (translate a file, run
quickLook)

dahmic77, 2024-10-29:
Yes, it works, if I start the command outside the /opt/gnssrefl main directory.