kristinemlarson/gnssrefl #320: ModuleNotFoundError: No module named 'gnssrefl.gpssnr'

dahmic77, 2024-10-29:
Hi, I have the same issue, than 'cheeseplam'. I followed your instruction under "Direct Python Install" (https://gnssrefl.readthedocs.io/en/latest/pages/README_install.html):
`pip install . `
`installexe linux64`    (Note: The EXE environment variable has to be set via .bashrc and also the EXE directory has to be created before this command work.)

Version gnssrefl-3.10.2 was installed at my Linux PC. Then I tried to run 'rinex2snr -h' command and got the same failure:
**ModuleNotFoundError: No module named 'gnssrefl.gpssnr'**

'gpssnr.f' is Fortran code. In a former version of gnssrefl a 'build' directory was created with Fortran binaries. But it seems to be that the installation steps above do not include a compilation of the Fortran code at the moment. Is the compilation of the Fortran code not necessary anymore?

_Originally posted by @dahmic77 in https://github.com/kristinemlarson/gnssrefl/issues/318#issuecomment-2445368287_

dahmic77, 2024-10-29:
I found now when this issue occurs. The failure occurs only if I start the 'rinex2snr -h' commando in the ./gnssrefl installation directory otherwise it works.

kristinemlarson, 2024-10-30:
I do not know why this isn't working or why the solution works. If anyone has better package management know-how, please contact @dahmic77. to me it looks strange that i have /opt/anaconda on the first line but /opt/gnssrefl on all the others, but this could easily be normal behavior.