kristinemlarson/gnssrefl #332: Differences in subdaily output files for multi-year runs

dnowacki-usgs, 2025-02-11:
gnssrefl                  3.10.6 on wsl, docker install (running via podman)

I have a station (I call it glv0) collecting NMEA data. I converted NMEA to SNR using nmea2snr and created results using gnssir. Then I run subdaily. subdaily executes successfully and creates plausible data for all scenarios below. There are, however, two issues: a missing plot, and filename inconsistencies.

# Missing plot
When I specify a single year, e.g.,
`subdaily glv0 2024 -extension refr_model_1` 
`subdaily glv0 2025 -extension refr_model_1` 

I get multiple diagnostic plots, including 5 rhdot plots: glv0_rhdot1.png through glv0_rhdot5.png, 

However, when I specify multiple years, e.g., `subdaily glv0 2024 -year_end 2025 -extension refr_model_1` only four rhdot plots are created. glv0_rhdot3.png is not generated.

# Filename inconsistency
Additionally, the file naming convention changes when `-year_end` is specified. For a single year, the withrhdot and withrhdotIF  files are named like:

glv0_2024_subdaily_edit.txt.withrhdot
glv0_2024_subdaily_edit.txt.withrhdotIF

But for multiple years, files are named like:

glv0_2024_2025.withrhdot
glv0_2024_2025.withrhdotIF 

As an aside, the single-year and multi-year files are nearly identical (for a given year), but with small (typically sub-cm) differences in columns 23, 24, 25 of the withrhdotIF file. But this issue is primarily about the missing diagnostic figure and inconsistency in filenames. Thanks.

kristinemlarson, 2025-02-12:
thanks for showing this to me. i will look into it.

kristinemlarson, 2025-02-13:
so I think the filename differences are a feature instead of a bug. since subdaily used to only allow a single year - and that year was put in the filename, how could i create a multi-year file that told the user that multiple years were there?  

as to why there would be small changes, i think i understand that.  the rhdot result is calculated from a fit to all the data. and the multi-year fit would be slightly different than the one that is only one year.  it should differ most at that year change.  

i think the missing figure is just plain laziness.  but i will double check.

kristinemlarson, 2025-02-13:
so i have gone ahead and added rhdot3 back in.  that plot should be created.  none of the rhdot plots have a year attached to them.

dnowacki-usgs, 2025-02-13:
Great, thanks for the quick response. Regarding 

> so I think the filename differences are a feature instead of a bug. since subdaily used to only allow a single year - and that year was put in the filename, how could i create a multi-year file that told the user that multiple years were there?

Totally agree on the multiple years being needed in the filename for multi-year output. I was pointing out that the single-year output has `_subdaily_edit.txt` in the filename, while the multi-year does not. It seemed like an inconsistency that might want to be addressed.

kristinemlarson, 2025-02-13:
i am still working to fix up the filename issues - cause those were very
inconsistent.  the long filenames are a bit embarrassing.
each major computation (rhdot, IF correction) resulted in a new column
being added because i was so worried about screwing up column 3
(where the original RH are stored).  so i ended up also making a new
filename and a new column.  the other extra column is just
because it allows you to easily back out the surface velocity.

the joke was on me when i found out a student in germany was running the
code to completion and then using column
3 anyway.

kristinemlarson, 2025-02-13:
so i am going to try to post this new code later this afternoon.
unfortunately i have been having problems with my docker
creation on github.  i think it is timing out - but who knows.  sometimes
it works and sometimes it does not.

if i get that to work and the master is all up to date with github, i will
subsequently be accepting a PR from a colleague in australia.
this change is for soil moisture users and will not impact you.

k

kristinemlarson, 2025-02-13:
ok, it has been posted successfully.  the code has two sections.  the first section is just simple stuff, like looking
for very large outliers and possibly windowing data for you.  it does not add columns, so there are no special filenames that have the beginning and ending year in them.  if i am not mistaken,  those one year files are then read by the second part of the code that DOES create new files (with withrhdot and withrhdot.IF in the file endings.   Closing for now.  let me know if there are problems with this version.  as i said, another version will be added later today to fix a bug in the soil moisture code (vwc_input).  either version should be fine for you.

closed 2025-02-13 by kristinemlarson