kristinemlarson/gnssrefl #335: .json vwc variables overwrite command line inputs

GeorgeTownsendd, 2025-02-14:
This command does not run as expected, because the minvalperday is not used properly:

```
vwc p038 2017 -minvalperday 1
```

The input (for all vwc variables) is overwritten in the set_parameters function in phase_functions.py:

```
    lsp = gnssir.read_json_file(station, '',noexit=True)
    # pick up values in json, if available
    if 'vwc_min_soil_texture' in lsp:
        tmin = lsp['vwc_min_soil_texture']
```

It reads from the json input file from that station. In my case this was 5:

```
    "vwc_minvalperday": 5,
```

Result: My vwc command executed as if minvalperday was set to 5

Expected behaviour: I would expect the command line argument to override the config value in the json.

kristinemlarson, 2025-02-14:
i'll change this to your suggestion.

kristinemlarson, 2025-02-14:
I would never advocate using one track for soil moisture.  definitely you could make 12 hour averages or 6 hour averages with the recent constellation (we did our early work when there were very few L2C transmitting satellites).  we did make 12 hour products for soil moisture, but we never bothered to publish that work.

kristinemlarson, 2025-02-14:
i think i can solve this by removing all the defaults in vwc_cl.py. Most of them are None but not all. If they are all None (though this is really only an issue for the ones that are stored in the json) I will have a way to tell that the user has set them from the command line.

kristinemlarson, 2025-02-16:
let me know how the new version works for you.

GeorgeTownsendd, 2025-02-16:
Works perfectly for me. Thankyou very much!

closed 2025-02-16 by GeorgeTownsendd

kristinemlarson, 2025-02-16:
thank you for telling me about the ndim parameter for numpy.loadtxt!