kristinemlarson/gnssrefl #336: Processing local dat

ashrafmousa07, 2025-02-17:
I am trying to process my own data.
I put the hatanaka compressed file in $REFL_CODE/2019/rinex/suez
I put the coordinate as a file "llh_local.txt"  in $REFL_CODE/input
The nav. file is "auto0050.19n" in the $ORBITS/2019/nav

I am running the GNSSREFL 3.10.8. 
I used the python install under linux64 bit
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.2 LTS
Release:	24.04
Codename:	noble

I tried the examples p041 and p042. Things goes well

Now I am trying to processes my own data.
I am trying to process my own data.
I put the hatanaka compressed file in $REFL_CODE/2019/rinex/suez
I put the coordinate as a file "llh_local.txt"  in $REFL_CODE/input
The nav. file is "auto0050.19n" in the $ORBITS/2019/nav

I ran the command  
-rinex2snr suez 2019 5 -nolook T

However this is what I get. 
Using these instructions  /home/ashraf/Downloads/gnssrefl-master/gnssrefl/refl_CODE/input/suez.json
Using snr value of  66
Station  suez  >>> RINEX 2.11
Debug mode or only analyzing one day of data. 
Feedback about the RINEX translation:  /home/ashraf/Downloads/gnssrefl-master/gnssrefl/refl_CODE/logs/suez/2019//005_translation.txt.gen
Log for Fortran translation :  /home/ashraf/Downloads/gnssrefl-master/gnssrefl/refl_CODE/logs/suez/2019//005_translation.txt
SNR file does not already exist. Which means I will try to make it.
Will first assume station  suez  year: 2019  doy: 5 is located here : /home/ashraf/Downloads/gnssrefl-master/gnssrefl
Did not find the plain observation file, so now trying other names/directories
Checking for the file in :  /home/ashraf/Downloads/gnssrefl-master/gnssrefl/refl_CODE/2019/rinex/suez/
You Chose the No Look Option, but did not provide the needed RINEX file.

Which mean that the program can not find the rinex file.

Please note that I also tried to put my data in the GNSSrefl dir, but still get he same error as above.

Could you please help

kristinemlarson, 2025-02-17:
Thank you for this description. I think the main issue missing  is that you have not told me the name of your RINEX file. If the file doesn't use the name the code is expecting, rinex2snr will not find the file.  Please post the name of your file.

kristinemlarson, 2025-02-18:
You can also try -mk T as an input. This will allow the station name to be all capitals. This option was originally written by someone who liked to use capitals.  though i cannot remember if the rest of the filename has to be capitals.  But give it a try.

ashrafmousa07, 2025-02-18:
Thank you so much for your kind concern
The station name is suez
I am using all names in small letters now
The file is suez0050.19o
Also suez0050.19d is available
Thanks in advance

ashrafmousa07, 2025-02-19:
Thanks alot for your kind help.
Now the program recognize the station file.
However, it does not recognize the orbit file.
orbit file is in $ORBITS/1019/nav/auto0050.19n
Is that write. why it ask for orbit on line!
here is the message:
gnssrefl version: 3.10.8 

Using these instructions  /home/ashraf/Downloads/gnssrefl-master/gnssrefl/REFL_CODE/input/hurg.json
Using snr value of  66
Station  hurg  >>> RINEX 2.11
You have invoked the Makan option
Debug mode or only analyzing one day of data. 
Feedback about the RINEX translation:  /home/ashraf/Downloads/gnssrefl-master/gnssrefl/REFL_CODE/logs/hurg/2019//005_translation.txt.gen
Log for Fortran translation :  /home/ashraf/Downloads/gnssrefl-master/gnssrefl/REFL_CODE/logs/hurg/2019//005_translation.txt
SNR file does not already exist. Which means I will try to make it.
Will first assume station  hurg  year: 2019  doy: 5 is located here : /home/ashraf/Downloads/gnssrefl-master/gnssrefl
hurg0050.19o hurg0050.19d
Will look for files in the working directory and  /home/ashraf/Downloads/gnssrefl-master/gnssrefl/REFL_CODE/rinex/hurg/2019/
Did not find the plain observation file, so now trying other names/directories
Checking for the file in :  /home/ashraf/Downloads/gnssrefl-master/gnssrefl/REFL_CODE/2019/rinex/hurg/
copying Hatanaka file to local area
Requested a GPS only nav file
Nav file exists online

No SNR file created - check error logs - locations were printed to screen

Thanks a lot

kristinemlarson, 2025-02-19:
No SNR file created - check error logs - locations were printed to screen

What does the error log say?

You can always use -orb gnss and avoid the issue of the nav file.    

I assume this is just a typo on the year?
orbit file is in $ORBITS/1019/nav/auto0050.19n

ashrafmousa07, 2025-02-23:
Thanks a lot.
I found that my file has no SNR record.
I will use the original file (bin. file) and recreat the rinex file with SNR data and try again
Cheers

closed 2025-03-03 by kristinemlarson