kristinemlarson/gnssrefl #339: Pip Install of gnssrefl failed

knelson2009, 2025-03-06:
I am having issues installing an updated version of gnssrefl on my lab's servers. It looks like it has something to do with the fortran executables, but I have not been able to trace it past that. The full log file also does not appear to exist at the directory provided. Please see the full output and error following the installation instructions below:

(base) [knelson12@stratus software]$ cd gnssrefl/
(base) [knelson12@stratus gnssrefl]$ ls
CHANGELOG.md        docs         meson.build     README_VEG.md         vendor
CITATION.cff        gnssrefl     notebooks       requirements-dev.txt
developer_notes.md  LICENSE      pyproject.toml  software_tests
Dockerfile          MANIFEST.in  README.md       test
(base) [knelson12@stratus gnssrefl]$ which gcc
/usr/local/bin/gcc
(base) [knelson12@stratus gnssrefl]$ which gfortran
/usr/local/bin/gfortran
(base) [knelson12@stratus gnssrefl]$ pip install .
Processing /cloud2/software/gnssrefl
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [12 lines of output]
      + meson setup /cloud2/software/gnssrefl /cloud2/software/gnssrefl/.mesonpy-whxcw1f8 -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md --native-file=/cloud2/software/gnssrefl/.mesonpy-whxcw1f8/meson-python-native-file.ini
      The Meson build system
      Version: 1.7.0
      Source dir: /cloud2/software/gnssrefl
      Build dir: /cloud2/software/gnssrefl/.mesonpy-whxcw1f8
      Build type: native build
      Project name: gnssrefl
      Project version: undefined
      
      ../meson.build:1:0: ERROR: Executables created by fortran compiler gfortran are not runnable.
      
      A full log can be found at /cloud2/software/gnssrefl/.mesonpy-whxcw1f8/meson-logs/meson-log.txt
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

My current version of gfortran is:
(base) [knelson12@stratus gnssrefl]$ gfortran --version
GNU Fortran (GCC) 4.9.4

My current version of gcc is:
base) [knelson12@stratus gnssrefl]$ gcc --version
gcc (GCC) 4.9.4

My server operating system information is as follows:
(base) [knelson12@stratus gnssrefl]$ cat /etc/os-release
NAME="Red Hat Enterprise Linux"
VERSION="8.7 (Ootpa)"
ID="rhel"
ID_LIKE="fedora"
VERSION_ID="8.7"
PLATFORM_ID="platform:el8"
PRETTY_NAME="Red Hat Enterprise Linux 8.7 (Ootpa)"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:redhat:enterprise_linux:8::baseos"
HOME_URL="https://www.redhat.com/"
DOCUMENTATION_URL="https://access.redhat.com/documentation/red_hat_enterprise_linux/8/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"

REDHAT_BUGZILLA_PRODUCT="Red Hat Enterprise Linux 8"
REDHAT_BUGZILLA_PRODUCT_VERSION=8.7
REDHAT_SUPPORT_PRODUCT="Red Hat Enterprise Linux"
REDHAT_SUPPORT_PRODUCT_VERSION="8.7"

Please let me know if there is a fix in progress, or if you need more information. Thank you for writing this software!

kristinemlarson, 2025-03-06:
> (base) ***@***.*** software]$ cd gnssrefl/
> (base) ***@***.*** gnssrefl]$ ls
> (base) ***@***.*** gnssrefl]$ which gcc
> (base) ***@***.*** gnssrefl]$ which gfortran
> (base) ***@***.*** gnssrefl]$ pip install .
> (base) ***@***.*** gnssrefl]$ gfortran --version
> base) ***@***.*** gnssrefl]$ gcc --version
> (base) ***@***.*** gnssrefl]$ cat /etc/os-release

so you were able to install gnssrefl before i changed to meson build?  it
looks like i did that for version 3.8.0
please confirm.

this is why we did it:
https://numpy.org/doc/stable/reference/distutils_status_migration.html

which version of python do you have installed?  i have mine installed with
3.9.
but i am running on a mac and

gfortran --version gives me

GNU Fortran (Homebrew GCC 14.1.0) 14.1.0
Copyright (C) 2024 Free Software Foundation, Inc.

gcc --version
Apple clang version 14.0.3 (clang-1403.0.22.14.1)
Target: arm64-apple-darwin22.6.0
Thread model: posix

i haven't used redhat in a long time so not familiar with the versions.
our docker is made with
Debian and python 3.10.

k.

knelson2009, 2025-03-06:
I am currently running Python 3.12.9 through miniconda3. The meson-log.txt file at the path you asked for is empty/does not exist.

kristinemlarson, 2025-03-06:
python 3.12.9 : well you are way ahead of me!

knelson2009, 2025-03-06:
For what it's worth, I've tried this with both pip and pip3. I also installed meson-python and got the same errors.

kristinemlarson, 2025-03-06:
i googled the error message

https://github.com/mesonbuild/meson/issues/11873

but i don't know enough to decipher your problem from it.

i will put out an email to the wider community to see if anyone is using
redhat.
i am pretty sure i got it to work with Ubuntu - but have to doublecheck on
that.

i'll look at what the docker build says ...

k.

kristinemlarson, 2025-03-06:
> I saw that the gfortran we have is much older. While you check with your
> colleagues, I will see if I can get gfortran and gcc to update and
> hopefully that solves the problem.
>
> I am not currently on the gnss-ir mailing list. I work more in GNSS-RO,
> but have begun branching out a bit to help with other projects.

ok. send me an email if you want to be added. k.

kristinemlarson, 2025-03-06:
Have you tried using the pypi version?  pip install gnssrefl.

knelson2009, 2025-03-06:
I have, but I receive the same errors. I suspect that the outdated compilers might be the problem. My colleague is trying to update them since I don't have admin access. Will let you know if that solves the problem.

kristinemlarson, 2025-03-17:
did this get solved?

knelson2009, 2025-04-23:
Hi Kristine, 

Unfortunately this still has not been resolved. I just got some admin access back to update various packages for the OS. I'm not sure why, but the RedHat package manager yum indicates that the most recent version of the gcc toolkit is gcc 4.9.4 and gfortran 4.9.4. I checked the gfortran distributions [here](https://gcc.gnu.org/wiki/GFortranDistros) and it looks like that is up to date for RedHat Fedora 21, which is what we are running. 

Are you sure that this software can build on RHEL OS? I was able to get gnssrefl to install correctly on my MacOS local machine but I am still having issues with installing on the servers. I have asked some IT folks what they might recommend or be able to help with it, so I will update when I can.

- Kevin

knelson2009, 2025-04-26:
Hi Kristine,

Good news! I was able to get gnssrefl installed on RHEL Fedora after working with my IT staff a bit. Turns out that you need to install gcc-toolset-14 first, then run the command "scl enable gcc-toolset-14 bash" (for bash shell). After that, a simple "pip install gnssrefl" did the job. Thank you for being so attentive to the issue!

closed 2025-04-26 by knelson2009