kristinemlarson/gnssrefl #341: Inconsistency between BDS 306/307 and S6/S7

Baizitang, 2025-03-25:
I am writing to report an issue I've encountered. I'm uncertain whether it's a problem with the description or the naming settings in the program. There is an inconsistency between the designations 306/307 and S6/S7. Through wavelength comparison, I've found that they are exactly the opposite.
For example, according to the expected wavelength - value relationship, the wavelength corresponding to 306 should match that of S6, but in my observations, the wavelength of 306 actually matches that of S7, and vice versa for 307.

[image]

[image]

This inconsistency might lead to incorrect data interpretation and potential errors in subsequent analyses or operations that rely on these designations.

I hope this issue can be investigated and resolved. If there's any additional information I can provide to assist in the troubleshooting process, please let me know.

Thank you.

Best regards,
Bai Zitang

kristinemlarson, 2025-03-27:
I have always struggled with how to code up the Chinese frequencies.  I appreciate your offer of help.  Right now I have been calling them 302 and 306.  But I need to go look at the frequencies defined in the code. I used to have 307 in there - but took it out at some point because I was not sure.

kristinemlarson, 2025-03-27:
Here is what I have in gps.py  But I would have to check against what I am calling these with respect to RINEX.  

#   beidou frequencies and wavelengths
    bei_L2 = 1561.098
    bei_L7 = 1207.14
    bei_L6 = 1268.52
    wbL2 = c/(bei_L2*1e6)
    # these values are defined in Rinex 3 
    wbL7 = c/(bei_L7*1e6)
    wbL6 = c/(bei_L6*1e6)

kristinemlarson, 2025-03-27:
this is in gnssir_cl.py
302,306,307 : BEIDOU B1, B3, B2 (not sure we do 307)

these are the beidou obs extracted from RINEX 3 files, where it keeps whatever one it finds first.  
gobblygook =  C:S2C,S7C,S6C,S2I,S7I,S6I,S2X,S6X,S7X'

i will do a quick test with 307.  i don't usually have files with it.

kristinemlarson, 2025-03-27:
Looks like I was not sure I was correct, so removed it, at least from quickLook,

_As far as I know, this is an illegal frequency for this software. Exiting.  307_

If i am using the wrong frequency (in gps.py) can you advise what to change it to?  If you don't know how to submit a PR, I can edit the code if you tell me what to change. And I can remove the "illegal frequency" comment.

Baizitang, 2025-03-28:
I believe there is no problem with your code. Just like defining wavelengths using carrier names for GPS and such, it should be sufficient to clearly write out the corresponding relationships in the manual.
If you need data from BeiDou for testing, I think the HKQT station in Hong Kong, China, should include S7I.

[image]

kristinemlarson, 2025-03-28:
looks to me like everything is working.  thanks much for bringing this up.

closed 2025-03-28 by kristinemlarson